mees/calvin #4: I get the following error when tring to clone from calvin_env

nikepupu, 2021-12-15:
**1. clone from calvin_env**
git clone  git@github.com:mees/calvin_env.git
Cloning into 'calvin_env'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

2. **directly following  the instruction**
git clone --recurse-submodules https://github.com/mees/calvin.git
Cloning into 'calvin'...
remote: Enumerating objects: 1360, done.
remote: Counting objects: 100% (1360/1360), done.
remote: Compressing objects: 100% (726/726), done.
remote: Total 1360 (delta 815), reused 1108 (delta 570), pack-reused 0
Receiving objects: 100% (1360/1360), 1.56 MiB | 6.71 MiB/s, done.
Resolving deltas: 100% (815/815), done.
Submodule 'calvin_env' (git@github.com:mees/calvin_env.git) registered for path 'calvin_env'
Cloning into '/mnt/research/calvin/calvin_env'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

mees, 2021-12-15:
Hi @nikepupu , thanks for trying out CALVIN!
I think the problem is that we used SSH to clone the submodules calvin_env and tacto. I guess you have not uploaded a SSH key to github and that is the reason why it failed to clone the submodules. I have changed the urls to use the HTTPS protocol, so now it should work for you. Can you try following the readme instructions with the latest commit and let us know if it solved your issue?

nikepupu, 2021-12-15:
Thanks a lot. The issue has been resolved.

closed 2021-12-15 by nikepupu

emrecanacikgoz, 2022-12-23:
The submodule issue is not related to ssh. I've faced the same issue both with https and ssh today. I've solved the issue by following:
```
cd parent-repo/
git submodule update --init --recursive
```
ref: https://stackoverflow.com/questions/47606539/cant-git-pull-a-submodule-when-git-clone-of-repo-wasnt-recursive
Hope it helps if someone else faces with similar issue..