mees/calvin #5: "InvalidGitRepositoryError" while running jupyter notebook

2000222, 2021-12-15:
Hi, I'd like to run RL_with_CALVIN.ipynb file in my locally established environment, but I meet the following issue while running the line "env = hydra.utils.instantiate(cfg.env)":
‘InvalidGitRepositoryError: Error instantiating 'calvin_env.envs.play_table_env.PlayTableSimEnv' 
Exactly I don't know how to solve it, so please give more help.  

PS. it seems that there exists some script issues while opening RL_with_CALVIN.ipynb.

ErickRosete, 2021-12-15:
Which cell do you run when this error appears? could you try it again and send me a screenshot of the the error please?

ErickRosete, 2021-12-15:
We updated the google colab href to redirect to the appropriate branch, please let us know if the issue is still present after these changes.

2000222, 2021-12-17:
Actually I'm running RL_with_CALVIN.ipynb in my local machine since in colab I obtain issues on "numpy.core.multiarray failed to import" but I fixed it locally, and I've finished the previous cell step by step as follows: 
[image]

[image]

[image]

[image]

[image]
 
I guess the problem is from calvin_env/utils/utils.py, but I'm not sure how to fix it. Thanks 💛

ErickRosete, 2021-12-17:
Thanks for your reply, now that we know that you are running on a local machine, it is possible that the problem was the installation / cloning of the repo.
Did you clone the repository before starting to run the notebook?  is there a folder named ".git/" inside the calvin_env folder?

Note: you should be able to run it in colab, by following the instructions. The line:
`!pip3 install -U numpy`
Is used to avoid the "numpy.core.multiarray failed to import" error that is present in colab, you could try running it again and restarting the runtime after the installation.

2000222, 2021-12-19:
Thanks a lot, I've solved the problem in Colab usage ~~ 😃

closed 2021-12-19 by 2000222