mees/calvin #6: I am trying to follow the instruction to run the code, however, after resolving some errors, there are some errors I cannot fix

nikepupu, 2021-12-16:
The command I exexuted :  
$ cd $CALVIN_ROOT/calvin_models/calvin_agent
$ python training.py

errors I got after fixing some errors. 
File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/utils.py", line 573, in _locate
    raise ImportError(f"Error loading module '{path}'") from e
ImportError: Error loading module 'lfp.utils.transforms.NormalizeVector'

I investigate the github, and I found out that this file is from here: which is missing 
[image]

python /home/hermannl/repos/learning_from_play/lfp/training.py  

In addition, I need to modify utils.transforms.py to get to this point. 

I modify ScaleImageTensor to this:
class NormalizeVector(object):
    """Normalize a tensor vector with mean and standard deviation."""

    def __init__(self, mean=0.0, std=1.0):
        **if isinstance(mean, float):
            mean = [mean]
        if isinstance(std, float):
            std = [std]**
        print("success")
        self.std = torch.Tensor(std)
        self.std[self.std == 0.0] = 1.0
        self.mean = torch.Tensor(mean)

    def __call__(self, tensor: torch.Tensor) -> torch.Tensor:
        assert isinstance(tensor, torch.Tensor)
        return (tensor - self.mean) / self.std

    def __repr__(self):
        return self.__class__.__name__ + "(mean={0}, std={1})".format(self.mean, self.std)

I also modify add depthNoise to this:
class AddDepthNoise(object):
    """Add multiplicative gamma noise to depth image.
    This is adapted from the DexNet 2.0 code.
    Their code: https://github.com/BerkeleyAutomation/gqcnn/blob/master/gqcnn/training/tf/trainer_tf.py"""

    def __init__(self, shape=1000.0, rate=1000.0):
        self.shape = torch.tensor(shape)
        self.rate = torch.tensor(rate)
        self.dist = torch.distributions.gamma.Gamma(torch.tensor(shape), torch.tensor(rate))

    def __call__(self, tensor: torch.Tensor) -> torch.Tensor:
        assert isinstance(tensor, torch.Tensor)
        multiplicative_noise = self.dist.sample()
        return multiplicative_noise * tensor

    def __repr__(self):
        # return self.__class__.__name__ + f"{self.shape=},{self.rate=},{self.dist=}"
        **return self.__class__.__name__ + "(self.shape={0}, self.rate={1}, self.dist={2})".format(self.shape, self.rate, self.dist)**

mees, 2021-12-17:
Hello @nikepupu ,
are you trying to train with a slurm cluster or directly? Because the hermannl repo is inside the slurm script directories, but in the beginning you say you try to run python training.py 
If you are trying to train directly, could you just post the log and the errors you get without any other modifications please?

nikepupu, 2021-12-17:
The first error is the following:
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/utils.py", line 585, in _locate
    import_module(mod)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 860, in get_code
  File "<frozen importlib._bootstrap_external>", line 791, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/mnt/research/calvin/calvin_models/calvin_agent/visualization/tsne_plot.py", line 85
    n := np.where(labels == -1)[0], replace=False, size=int(len(n) * self.plot_percentage)
      ^
SyntaxError: invalid syntax

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "training.py", line 136, in <module>
    wrap_train(conf)
  File "training.py", line 64, in wrap_train
    train()
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/main.py", line 52, in decorated_main
    config_name=config_name,
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/utils.py", line 378, in _run_hydra
    lambda: hydra.run(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/utils.py", line 214, in run_and_report
    raise ex
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/utils.py", line 211, in run_and_report
    return func()
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/utils.py", line 381, in <lambda>
    overrides=args.overrides,
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/hydra.py", line 111, in run
    _ = ret.return_value
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/core/utils.py", line 233, in return_value
    raise self._return_value
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/core/utils.py", line 160, in run_job
    ret.return_value = task_function(task_cfg)
  File "training.py", line 40, in train
    callbacks = setup_callbacks(cfg.callbacks)
  File "training.py", line 68, in setup_callbacks
    callbacks = [hydra.utils.instantiate(cb) for cb in callbacks_cfg.values()]
  File "training.py", line 68, in <listcomp>
    callbacks = [hydra.utils.instantiate(cb) for cb in callbacks_cfg.values()]
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 180, in instantiate
    return instantiate_node(config, *args, recursive=_recursive_, convert=_convert_)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 240, in instantiate_node
    _target_ = _resolve_target(node.get(_Keys.TARGET))
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 104, in _resolve_target
    return _locate(target)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.7/site-packages/hydra/_internal/utils.py", line 589, in _locate
    ) from e
ImportError: Encountered error: `invalid syntax (tsne_plot.py, line 85)` when loading module 'calvin_agent.visualization.tsne_plot.TSNEPlot'

I am using Anaconda on Ubuntu 18.04 with python3.7 with a RTX 3090 if that helps

nikepupu, 2021-12-18:
After switching to python3.8, the above issue can be resolved. However, the next issue still exists, which is the modification I mentioned in the first post. 

Using 16bit native Automatic Mixed Precision (AMP)
GPU available: True, used: True
TPU available: False, using: 0 TPU cores
IPU available: False, using: 0 IPUs
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 62, in _call_target
    return _target_(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/utils/transforms.py", line 23, in __init__
    self.std = torch.Tensor(std)
TypeError: new(): data must be a sequence (got float)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "training.py", line 62, in train
    trainer.fit(model, datamodule=datamodule)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 737, in fit
    self._call_and_handle_interrupt(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 682, in _call_and_handle_interrupt
    return trainer_fn(*args, **kwargs)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 772, in _fit_impl
    self._run(model, ckpt_path=ckpt_path)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1133, in _run
    self._call_setup_hook()  # allow user to setup lightning_module in accelerator environment
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1433, in _call_setup_hook
    self.datamodule.setup(stage=fn)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/core/datamodule.py", line 474, in wrapped_fn
    fn(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 62, in setup
    self.train_transforms = {
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 63, in <dictcomp>
    cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 63, in <listcomp>
    cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 180, in instantiate
    return instantiate_node(config, *args, recursive=_recursive_, convert=_convert_)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 249, in instantiate_node
    return _call_target(_target_, *args, **kwargs)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 64, in _call_target
    raise type(e)(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 62, in _call_target
    return _target_(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/utils/transforms.py", line 23, in __init__
    self.std = torch.Tensor(std)
TypeError: Error instantiating 'calvin_agent.utils.transforms.NormalizeVector' : new(): data must be a sequence (got float)

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

mees, 2021-12-19:
thanks for the report @nikepupu . I have updated the instructions and made a minor fix. The problem is that the path for the dataset was being incorrectly set. I have updated the default to dataset/task_D_D, could you try starting the training with and without overriding the argument?
So once:
`python training.py`
and once:
`python training.py datamodule.root_data_dir=/path/to/dataset/`

closed 2021-12-19 by mees

nikepupu, 2021-12-20:
This is what I am getting:
 python training.py
/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config.yaml': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
Global seed set to 42
[2021-12-19 17:34:59,653][calvin_agent.models.decoders.logistic_policy_network][INFO] - Loaded action bounds from /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/training/statistics.yaml
[2021-12-19 17:34:59,673][__main__][INFO] - Training with the following config:
callbacks:
  rollout:
    tasks:
      _target_: calvin_env.envs.tasks.Tasks
      tasks:
        rotate_red_block_right:
        - rotate_object
        - block_red
        - -60
        rotate_red_block_left:
        - rotate_object
        - block_red
        - 60
        rotate_blue_block_right:
        - rotate_object
        - block_blue
        - -60
        rotate_blue_block_left:
        - rotate_object
        - block_blue
        - 60
        rotate_pink_block_right:
        - rotate_object
        - block_pink
        - -60
        rotate_pink_block_left:
        - rotate_object
        - block_pink
        - 60
        push_red_block_right:
        - push_object
        - block_red
        - 0.1
        - 0
        push_red_block_left:
        - push_object
        - block_red
        - -0.1
        - 0
        push_blue_block_right:
        - push_object
        - block_blue
        - 0.1
        - 0
        push_blue_block_left:
        - push_object
        - block_blue
        - -0.1
        - 0
        push_pink_block_right:
        - push_object
        - block_pink
        - 0.1
        - 0
        push_pink_block_left:
        - push_object
        - block_pink
        - -0.1
        - 0
        move_slider_left:
        - move_door_rel
        - base__slide
        - 0.15
[... 499 lines cut ...]
    self.train_transforms = {
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 63, in <dictcomp>
    cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 63, in <listcomp>
    cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 180, in instantiate
    return instantiate_node(config, *args, recursive=_recursive_, convert=_convert_)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 249, in instantiate_node
    return _call_target(_target_, *args, **kwargs)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 64, in _call_target
    raise type(e)(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 62, in _call_target
    return _target_(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/utils/transforms.py", line 23, in __init__
    self.std = torch.Tensor(std)
TypeError: Error instantiating 'calvin_agent.utils.transforms.NormalizeVector' : new(): data must be a sequence (got float)

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

wandb: Waiting for W&B process to finish, PID 7029... (failed 1). Press ctrl-c to abort syncing.
wandb:                                                                                
wandb: Synced 6 W&B file(s), 0 media file(s), 0 artifact file(s) and 0 other file(s)
wandb: Synced 2021-12-19/17-34-59: https://wandb.ai/nikepupu/multi_play/runs/2021-12-19_17-34-59
wandb: Find logs at: ./wandb/run-20211219_173500-2021-12-19_17-34-59/logs/debug.log
wandb:

nikepupu, 2021-12-20:
I can modify this torch.Tensor to torch.tensor like other functions in this file to get rid of this error :

class NormalizeVector(object):
    """Normalize a tensor vector with mean and standard deviation."""

    def __init__(self, mean=0.0, std=1.0):
        self.std = torch.Tensor(std)
        self.std[self.std == 0.0] = 1.0
        self.mean = torch.Tensor(mean)

    def __call__(self, tensor: torch.Tensor) -> torch.Tensor:
        assert isinstance(tensor, torch.Tensor)
        return (tensor - self.mean) / self.std

    def __repr__(self):
        return self.__class__.__name__ + "(mean={0}, std={1})".format(self.mean, self.std)

However, the next error still exits, which is the error in the first post:

Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/utils.py", line 570, in _locate
    module = import_module(mod)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1011, in _gcd_import
  File "<frozen importlib._bootstrap>", line 950, in _sanity_check
ValueError: Empty module name

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "training.py", line 62, in train
    trainer.fit(model, datamodule=datamodule)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 737, in fit
    self._call_and_handle_interrupt(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 682, in _call_and_handle_interrupt
    return trainer_fn(*args, **kwargs)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 772, in _fit_impl
    self._run(model, ckpt_path=ckpt_path)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1133, in _run
    self._call_setup_hook()  # allow user to setup lightning_module in accelerator environment
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1433, in _call_setup_hook
    self.datamodule.setup(stage=fn)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/core/datamodule.py", line 474, in wrapped_fn
    fn(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 62, in setup
    self.train_transforms = {
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 63, in <dictcomp>
    cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 63, in <listcomp>
    cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 180, in instantiate
    return instantiate_node(config, *args, recursive=_recursive_, convert=_convert_)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 240, in instantiate_node
    _target_ = _resolve_target(node.get(_Keys.TARGET))
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 104, in _resolve_target
    return _locate(target)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/utils.py", line 573, in _locate
    raise ImportError(f"Error loading module '{path}'") from e
ImportError: Error loading module 'lfp.utils.transforms.NormalizeVector'

source of error
File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 63, in <dictcomp>
    cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train

nikepupu, 2021-12-20:
I think this is because you have a statistics.yaml file under the training folder and that file requires lfp.utils.transforms.NormalizeVector
[image]

I can get rid of this error by modifying statistics.yaml in both training and evaluation folder.

However, there is a new error showing up 

trying to load lang data from:  /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/training/lang_annotations/auto_lang_ann.npy
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 62, in _call_target
    return _target_(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/npz_dataset.py", line 43, in __init__
    ) = self.load_file_indices_lang(self.abs_datasets_dir)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/npz_dataset.py", line 149, in load_file_indices_lang
    assert end_idx >= self.max_window_size
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "training.py", line 62, in train
    trainer.fit(model, datamodule=datamodule)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 737, in fit
    self._call_and_handle_interrupt(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 682, in _call_and_handle_interrupt
    return trainer_fn(*args, **kwargs)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 772, in _fit_impl
    self._run(model, ckpt_path=ckpt_path)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1133, in _run
    self._call_setup_hook()  # allow user to setup lightning_module in accelerator environment
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1433, in _call_setup_hook
    self.datamodule.setup(stage=fn)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/core/datamodule.py", line 474, in wrapped_fn
    fn(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 73, in setup
    train_dataset = hydra.utils.instantiate(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 180, in instantiate
    return instantiate_node(config, *args, recursive=_recursive_, convert=_convert_)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 249, in instantiate_node
    return _call_target(_target_, *args, **kwargs)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 64, in _call_target
    raise type(e)(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 62, in _call_target
    return _target_(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/npz_dataset.py", line 43, in __init__
    ) = self.load_file_indices_lang(self.abs_datasets_dir)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/npz_dataset.py", line 149, in load_file_indices_lang
    assert end_idx >= self.max_window_size
AssertionError: Error instantiating 'calvin_agent.datasets.npz_dataset.NpzDataset' : 

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

nikepupu, 2021-12-20:
After removing the assertion, I encountered another error

Using 16bit native Automatic Mixed Precision (AMP)
GPU available: True, used: True
TPU available: False, using: 0 TPU cores
IPU available: False, using: 0 IPUs
/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/core/datamodule.py:121: LightningDeprecationWarning: DataModule property `train_transforms` was deprecated in v1.5 and will be removed in v1.7.
  rank_zero_deprecation(
/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/core/datamodule.py:140: LightningDeprecationWarning: DataModule property `val_transforms` was deprecated in v1.5 and will be removed in v1.7.
  rank_zero_deprecation(
/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/core/datamodule.py:114: LightningDeprecationWarning: DataModule property `train_transforms` was deprecated in v1.5 and will be removed in v1.7.
  rank_zero_deprecation(
/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/core/datamodule.py:133: LightningDeprecationWarning: DataModule property `val_transforms` was deprecated in v1.5 and will be removed in v1.7.
  rank_zero_deprecation(
[2021-12-19 21:46:20,396][calvin_agent.datasets.base_dataset][INFO] - loading dataset at /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/training
[2021-12-19 21:46:20,396][calvin_agent.datasets.base_dataset][INFO] - finished loading dataset
[2021-12-19 21:46:20,410][calvin_agent.datasets.npz_dataset][INFO] - Found "ep_start_end_ids.npy" with 1 episodes.
self.training_dir  /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/training
[2021-12-19 21:46:20,434][calvin_agent.datasets.base_dataset][INFO] - loading dataset at /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/validation
[2021-12-19 21:46:20,434][calvin_agent.datasets.base_dataset][INFO] - finished loading dataset
[2021-12-19 21:46:20,436][calvin_agent.datasets.npz_dataset][INFO] - Found "ep_start_end_ids.npy" with 1 episodes.
train_dataset <calvin_agent.datasets.npz_dataset.NpzDataset object at 0x7f18f197b610>
[2021-12-19 21:46:20,461][calvin_agent.datasets.base_dataset][INFO] - loading dataset at /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/training
[2021-12-19 21:46:20,461][calvin_agent.datasets.base_dataset][INFO] - finished loading dataset
trying to load lang data from:  /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/training/lang_annotations/auto_lang_ann.npy
self.training_dir  /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/training
[2021-12-19 21:46:20,475][calvin_agent.datasets.base_dataset][INFO] - loading dataset at /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/validation
[2021-12-19 21:46:20,475][calvin_agent.datasets.base_dataset][INFO] - finished loading dataset
trying to load lang data from:  /mnt/research/calvin/calvin_models/calvin_agent/dataset/task_D_D/validation/lang_annotations/auto_lang_ann.npy
train_dataset <calvin_agent.datasets.npz_dataset.NpzDataset object at 0x7f18f1984280>
Error executing job with overrides: []
Traceback (most recent call last):
  File "training.py", line 62, in train
    trainer.fit(model, datamodule=datamodule)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 737, in fit
    self._call_and_handle_interrupt(
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 682, in _call_and_handle_interrupt
    return trainer_fn(*args, **kwargs)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 772, in _fit_impl
    self._run(model, ckpt_path=ckpt_path)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1133, in _run
    self._call_setup_hook()  # allow user to setup lightning_module in accelerator environment
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1433, in _call_setup_hook
    self.datamodule.setup(stage=fn)
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/pytorch_lightning/core/datamodule.py", line 474, in wrapped_fn
    fn(*args, **kwargs)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 79, in setup
    train_sampler = get_sampler(train_dataset, shuffle=True)
  File "/mnt/research/calvin/calvin_models/calvin_agent/datasets/play_data_module.py", line 119, in get_sampler
    return RandomSampler(dataset, generator=torch.Generator().manual_seed(int(os.environ["PL_GLOBAL_SEED"])))  # type: ignore
  File "/home/nikepupu/anaconda3/envs/calvin/lib/python3.8/site-packages/torch/utils/data/sampler.py", line 102, in __init__
    raise ValueError("num_samples should be a positive integer "
ValueError: num_samples should be a positive integer value, but got num_samples=0

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

It seems this is because RandomSampler requires to pass in an argument of num_samples
[image]

[image]

mees, 2021-12-20:
Thanks for the logs. This is a bit weird because I checked the statistics inside the zip files and the paths for the statistics.yaml appear to be correct. Did you download the dataset recently? And do you get the same errors when you start the training by specifying the dataset with `python training.py datamodule.root_data_dir=/path/to/dataset/`
Also note that we tested the code with pytorch lightning 1.4.9, you might need to make changes if you want to user PL > 1.5 as you log indicates.

mees, 2021-12-20:
Just to clarify, you downloaded the dataset as specified in the readme like this?
$ cd $CALVIN_ROOT/dataset
$ sh download_data.sh D | ABC | ABCD

nikepupu, 2021-12-20:
I only did sh download_data.sh D

mees, 2021-12-20:
Okey, my suspicion is that you are not loading the correct dataset. In the code, we download a minimal dataset of 50 steps, which we used to run CI tests during development, when no dataset path is given and the default location is empty.
https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/datasets/play_data_module.py#L56

nikepupu, 2021-12-20:
I see, when I run my code the first time., I do observe it's downloading something.

mees, 2021-12-20:
Yes, I will fix that. But you should try running `python training.py datamodule.root_data_dir=/path/to/dataset/` where the dataset i s the path where you downloaded the dataset with the download script. Does this work for you?

mees, 2021-12-20:
I have fixed that problem, can you try again please? `python training.py datamodule.root_data_dir=/path/to/dataset/`

nikepupu, 2021-12-21:
It's working now. Thanks

commit 23ce311b (2021-12-19): fix #6  update instructions, set default root_data_dir