mees/calvin #9: Physics of the robot arm is odd

aliang8, 2022-01-07:
I am using the SlideEnv provided in the tutorial jupyter notebook for evaluating a trained RL policy. In some of my rollouts, I noticed in the static camera RGB image that there is an weird piece sticking out of the robot's arm. In the video I attached, that additional piece seems to be manipulating the sliding drawer. I also notice that this thing protuding from the arm in the picture of Env B from the paper. Is this supposed to be part of the environment. Thanks!

https://user-images.githubusercontent.com/16139634/148593615-9466f649-97f2-45cb-a198-9d15c43f0372.mp4

mees, 2022-01-07:
Thanks for trying out CALVIN @aliang8 !
Wow, that's a definitely cool novel way of sliding the drawer we have not observed before! :open_mouth:
The floating block you mention is the, not so realistically depicted, gripper camera. If you don't need the gripper camera you can remove it or allow collisions for it here https://github.com/mees/calvin_env/blob/main/data/franka_panda/panda.urdf#L346

lukashermann, 2022-01-08:
@mees I think we should disable collisions completely for the gripper camera, what do you think?

mees, 2022-01-08:
@lukashermann I agree.

lukashermann, 2022-01-11:
Collisions are now disabled.

closed 2022-01-11 by lukashermann