mees/calvin #10: Reseting env to state from dataset

aliang8, 2022-01-10:
Hi,

I'm trying to generating skill id / language annotations for the unlabeled frames in the dataset. I was thinking of using the `reset_from_storage` method in the environment class to reset to a state from the dataset and using the task checker to check for task success. However, the reset function requires a serialized version of the env/robot state which is not provided. Is there a way I could reset the env from offline data or is there another way for me to get skill annotations for the entire dataset?

Thanks!

aliang8, 2022-01-10:
Never mind! I could use: `env.reset(scene_obs=data["scene_obs"], robot_obs=data["robot_obs"])`. Thanks anyways!

lukashermann, 2022-01-10:
Yes, we use `reset_from_storage` only with the raw data saved during teleoperation.

closed 2022-01-10 by lukashermann

mees, 2022-01-11:
Hi @aliang8 , to label all the dataset you need to adjust the eps parameter for the  [automatic_lang_annotator_mp](https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/utils/automatic_lang_annotator_mp.py) script [here](https://github.com/mees/calvin/blob/main/calvin_models/conf/lang_ann.yaml#L18)

lukashermann, 2022-01-11:
Additionally, to get a feeling for the task detector in an interactive way, have a look at this [script](https://github.com/mees/calvin_env/blob/main/calvin_env/scripts/check_tasks.py).

You can navigate through the dataset with A and D keys and run the task detector with the R-key. There is no documentation yet because it was mainly used for debugging purposes.

aliang8, 2022-01-11:
Thanks @mees and @lukashermann for the replies. I ran the automatic_lang_annotator script and changed the eps value. However, the resulting annotations of skills are very unbalanced. Is this normal and how could I get the skill annotations to be more balanced? Thanks!

[image]

lukashermann, 2022-01-12:
Can you send the exact command that you used to generate the annotations? 
There must be something wrong, because with `eps =0.01` you get this result:

`Counter({'move_slider_left': 151,
         'turn_off_lightbulb': 152,
         'turn_on_led': 150,
         'turn_on_lightbulb': 150,
         'open_drawer': 152,
         'place_in_drawer': 152,
         'move_slider_right': 152,
         'lift_blue_block_table': 152,
         'turn_off_led': 150,
         'push_into_drawer': 150,
         'rotate_red_block_left': 150,
         'close_drawer': 151,
         'lift_red_block_drawer': 151,
         'unstack_block': 151,
         'lift_red_block_slider': 151,
         'lift_red_block_table': 150,
         'rotate_pink_block_right': 150,
         'rotate_pink_block_left': 151,
         'lift_pink_block_slider': 151,
         'lift_pink_block_table': 151,
         'place_in_slider': 151,
         'lift_blue_block_slider': 154,
         'lift_pink_block_drawer': 150,
         'rotate_red_block_right': 150,
         'lift_blue_block_drawer': 150,
         'stack_block': 151,
         'push_pink_block_left': 150,
         'rotate_blue_block_left': 150,
         'push_red_block_left': 150,
         'rotate_blue_block_right': 150,
         'push_blue_block_right': 150,
         'push_pink_block_right': 150,
         'push_red_block_right': 150,
         'push_blue_block_left': 150})
`

aliang8, 2022-01-13:
Also I find that with some tasks like "rotate_blue_block_right", a lot of the annotated intervals are duplicates / overlaps of each other. It says "rotate_blue_block_right" has 150 annotations, but actually there are only 29 distinct ones.