mees/calvin #11: Errors with EGL

zhaozj89, 2022-01-16:
Thanks for this work. When I followed the readme and ran `python training.py datamodule.root_data_dir=/path/to/dataset/`, it reported an error as `Segmentation fault (core dumped)` when loading EGL plugin in `calvin_env`.

I use a Ubuntu 16.04 sever with a Nvidia 2080Ti card, and the driver is `nvidia-container-runtime 3.5.0-1` and cuda is 11.2. I have searched the Internet for a while, such as installing mesa as`sudo apt-get install libglfw3-dev libgles2-mesa-dev`, but still did not work. 

I would like to inquire if you know how to enable EGL with my hardware setting and what is the function of EGL, for displaying the robot?

By the way, what is the time of training the baseline on the three datasets provided, i.e., task_D.zip, task_ABC_D.zip, task_ABCD_D.zip?

Thanks very much.

mees, 2022-01-17:
Hi @zhaozj89 , thanks for trying out CALVIN!
what pybullet version are you using? Older pybullet versions have spotty EGL support, so maybe installing the latest version might help. We use EGL to move the bullet rendering from cpu (which is the default) to gpu, which is much faster. This way, we can also do rollouts during the training of the agent to track its performance. By changing from cpu to gpu, the rendered textures change slightly, so be aware of this if you plan on testing pretrained models.
Regarding training, we followed the hyperparameters from Lynch et al. as closely as possible (batch size etc.). Training the agent with visual observations from the static camera for the single environment took around 3-4 days on 8 Nvidia 2080Ti cards.
If you want to just play around with CALVIN without learning from pixels and have faster training times, you could use the scene observations as input (position and orientation of the objects, state of the drawers etc.) instead of the visual observations.

zhaozj89, 2022-01-17:
Thanks very much for the reply. For the EGL issue, I found out that it may be my nvidia packages are not installed correctly. Anyway, thanks for this work!

closed 2022-01-17 by zhaozj89

lukashermann, 2022-01-17:
To add a bit more context for people with EGL problems:
PyBullet only recently added an option to select which GPU to use for rendering when using EGL (fix was commited in 3c4cb80
on Oct 22, 2021, see
https://github.com/bulletphysics/bullet3/blob/master/examples/OpenGLWindow/EGLOpenGLWindow.cpp#L134 )
So if you have an old version of PyBullet, there is no way to choose the GPU, which can lead to problems on cluster nodes with multiple GPUs, because all instances would be placed on the same GPU, slowing down the rendering and potentially leading to OOM erros. The fix introduced an environment variable EGL_VISIBLE_DEVICES (similar to CUDA_VISIBLE_DEVICES) which lets you specify the GPU device to render on.

However, there is one catch: On some machines, the device ids of CUDA and EGL do not match (e.g. CUDA device 0 could be EGL device 3). We automatically handle this in our wrapper in calvin_env and find the corresponding egl device id, so you don't have to set EGL_VISIBLE_DEVICES yourself, see https://github.com/mees/calvin_env/blob/main/calvin_env/envs/play_lmp_wrapper.py#L31.