mees/calvin #13: Stuck at beginning of training

dennisushi, 2022-03-17:
Hi!

Running the baseline training gets stuck at the very beginning. Do you have any clue why that might be? Is it normal for iterations to take 23.91s/it? There is no error. 

```bash
root_data_dir: /media/dennisushi/DREVO-P1/DATA/calvin/task_D_D/task_A_A
...
slurm: false
...
[2022-03-17 11:16:03,707][__main__][INFO] - * CUDA:
	- GPU:
		- GeForce RTX 3080
		- GeForce RTX 3080
	- available:         True
	- version:           11.1
* Packages:
	- numpy:             1.21.2
	- pyTorch_debug:     False
	- pyTorch_version:   1.12.0.dev20220224+cu111
	- pytorch-lightning: 1.4.9
	- tqdm:              4.63.0
...
...
[2022-03-17 11:16:22,988][calvin_agent.models.play_lmp][INFO] - Finished validation epoch 0
Global seed set to 42                                                                                                
Epoch 0:   0%|                                                                   | 0/19063 [00:00<00:04, 3979.42it/s][2022-03-17 11:16:23,004][calvin_agent.models.play_lmp][INFO] - Start training epoch 0
Epoch 0:   0%|                                          | 3/19063 [01:35<126:35:39, 23.91s/it, loss=42.1, v_num=6-02]
```

mees, 2022-03-17:
Hey @dennisushi thanks for trying out CALVIN!
Could you send us the exact command you sing to instantiate the training? Are you training on 2 GPUs? What is your batch size?
No, it should be much faster.

lukashermann, 2022-03-18:
this `root_data_dir` seems weird, why do you have `task_A_A` inside `task_D_D`? Our download script should rename the unpacked folder (https://github.com/mees/calvin/blob/main/dataset/download_data.sh#L10). Sorry for this confusion, the change of naming convention is a bit unfortunate.

dennisushi, 2022-03-18:
Hi, thanks for getting back to me so quickly.

@lukashermann I downloaded the data using `sh download_data.sh D`, I did not move any folders manually, although I had to stop in the middle of the unzipping due to lack of memory, clear some space and redo it. `mv` moves the folder inside the second, if the second already exists - which I guess was the case due to the interrupted unzipping. I just moved all of its contents to task_D_D - but this wasn't the issue, since task_A_A had all training/validation examples in it. 

@mees  the command I am using is `python3 training.py datamodule.root_data_dir=/media/dennisushi/DREVO-P1/DATA/calvin/task_D_D/task_A_A`, it seems to be using both GPUs according to the output `GPU available: True, used: True` and `CUDA_VISIBLE_DEVICES: [0,1]`. I have not made any changes to the code or any configuration. Simply installed, changed the torch to the nightly version and ran the command.

mees, 2022-03-18:
So by default only a single GPU is [used](https://github.com/mees/calvin/blob/main/calvin_models/conf/trainer/play_trainer.yaml#L1), to train on both GPUs override this parameter as specified in the readme with trainer.gpus=-1 (trains DDP with all visible GPUs)
However, I don't think that will change much. It would be great if you could try training with the specified Pytorch version, because I don't think that Lightning 1.4.9 supports Pytorch 1.2. You could also upgrade Lightning to continue using Pytorch 1.2, but some minor code changes might be necessary for this. Finally, you can also try to track GPU memory usage, to see if maybe you need to reduce the default batch size a bit.

lukashermann, 2022-03-18:
Can you try running it without the callbacks? Add this to the cmd line arguments `~callbacks/rollout ~callbacks/rollout_lh callbacks/checkpoint=val_action`

dennisushi, 2022-03-18:
The batch change of both vision and language improved it slightly but it was still getting stuck (until step 400). The callback comment improved that further (until step 680 in 2 minutes) but still gets stuck. Using `trainer.gpus=-1` kills the run as soon as training epoch 0 starts, which is weird. I guess this is something I should run on our cluster instead of the PC. Thanks a lot for the help.

mees, 2022-03-18:
> The batch change of both vision and language improved it slightly but it was still getting stuck (until step 400). The callback comment improved that further (until step 680 in 2 minutes) but still [...]

If its getting killed with multi-gpu training, does the error log give any info on why its terminating?

lukashermann, 2022-03-21:
> Actually, I noticed that it always reaches the previous longest step quickly and then adds a few slowly before stopping. Hence why each attempt seems faster than the previous. Maybe it's using some checkpoint to reach it? I don't know

This is probably because there is some caching involved. The random sampling of data points is always in the same order, unless you change the seed.

The code should run on a normal workstation, especially when it has an RTX 3080.
How many cpu cores do you have? You can try to run it with less data loading workers. For that, just add the flag `datamodule.num_workers=2` or even `datamodule.num_workers=1`. Our default value is 4, but since we have two datasets for vision and language, that gives you 8 workers. I remember that I had some trainings that froze in the past when I was using too many workers.

dennisushi, 2022-03-23:
Hi,

I had left it for a few days - the GPU killing doesn't happen today so I can't reproduce it. I have 20 CPU cores. 
I was already setting it to 1 worker, so no change with this.
One potential explanation could be that the data is on an external hard drive? It has not been an issue in the past but perhaps the I/O is trying to read many .npy files at the same time? I will investigate

Regards,
Dennis

closed 2022-03-31 by mees