mees/calvin #15: May a have a small part of data to get started?

yizhouzhao, 2022-05-07:
Dear authors, 

Thanks for this great work. It inspires me a lot.

However, the dataset size is too overwhelming to my desktop to get started. 

May I have a small fraction of data/demo/performance test to just run the pre-trained model?

mees, 2022-05-07:
Thanks @yizhouzhao, we will release soon a mini dataset for a quick start. @lukashermann maybe we should include instructions for "ask me anything" rollouts with a trained model?

lukashermann, 2022-05-13:
@yizhouzhao thanks for your request, we uploaded a small debug dataset of 1.3 GB. Check the updated information in the dataset [README](https://github.com/mees/calvin/tree/main/dataset). If you have further questions about the dataset, don't hesitate to ask! :)

closed 2022-05-13 by lukashermann