mees/calvin #23: The proportion of the recorded robot interaction data with language instructions

geyuying, 2022-08-01:
Hi，

Thanks for your excellent benchmark!

I have a question regarding the proportion of the recorded robot interaction data with language instructions.

The CALVIN paper says thet "we annotate only 1% of the recorded robot interaction data with language instructions." 

After I download the dataset "task_D_D", "ep_start_end_ids.npy" under the training folder records 512046 unique episodes (saved as .npz file). Under "training/lang_annotations" folder, "auto_lang_ann.npy" records 303794 episodes with 192607 unique episodes. In this way, it seems that 192607 episodes are annotated with language instructions among all the 512046 episodes in the training set. The proportion is 192607/512046 , different from  1% in the paper.

If my analysis is correct, why is the the proportion of the recorded robot interaction data with language instructions in the dataset different from that in paper?

Looking forward to your reply.

lukashermann, 2022-08-02:
I think the problem here is the definition of what is a unique episode. The naming pattern in the dataset is very unfortunately chosen und due to legacy, each frames (which are single steps) are called episodes. In the paper, what we call an episode, is just a trajectory (or a window) of consecutive steps from the play data. This means, that one unique frame can be present in several episodes, since we are basically sliding a window over the stream of play data to generate all episodes. Thus when labeling language, the same frame can be contained in multiple episodes. When we say we labeled 1% of the dataset, this means that we labeled 1% of the episodes (sequence windows). 

To our understanding, this definition is consistent with the original definition of Lynch and Sermanet in their paper [Language Conditioned Imitation Learning over Unstructured Data](https://arxiv.org/pdf/2005.07648.pdf) (Appendix A and B).

closed 2022-08-02 by lukashermann

geyuying, 2022-08-02:
Thanks for your detailed explanation!

The length of the self.episode_lookup (defined in [disk_dataset.py)](https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/datasets/disk_dataset.py) seems to correspond to the length of the episodes (sequence windows) that you mentioned.  

In the "task_D_D" training set, I print out the length of the self.episode_lookup. "vis" has 511457 episodes and "lang" has 206438 episodes.  It seems that 40% episodes are labeled with languages.

geyuying, 2022-08-02:
Yes! That solved my question. Thanks again for your clear and detailed explanation.

By the way, what is the respective role of human annotators and the automatic labeling tool?

lukashermann, 2022-08-02:
We did not use human annotators, I just wanted to give an explanation for why a longer sequence length is necessary for labeling (also if we had used human annotators).

kahnchana, 2025-01-13:
Another clarification on this. 

When going through the data (task_D_D), I notice 5124 language annotations in `auto_lang_ann.npy` (i.e. the length of `auto_lang_ann['language']['ann']`). These 5124 captions contain 389 unique ones. 

Is the total unique action sequences in task_D_D 5124? And the 206438 is obtained by sliding window style sub-sampling? 

Also, does anyone know these numbers for A B C envs?