mees/calvin #24: `scripts/automatic_lang_annotator_mp.py` appears to be broken

ezhang7423, 2022-08-04:
The configuration file that this script uses, `lang_ann.yml`, does not appear to have two required keys: `rollout_sentences` and `annotations`. To replicate this, one can simply try running the script.

ezhang7423, 2022-08-04:
Command: 
`python3 hulc/utils/automatic_lang_annotator_mp.py datamodule.root_data_dir=/data2/eddie/calvin/task_D_D datamodule/datasets=vision_lang +rollout_sentences=new_playtable`
Output:
```
[2022-08-04 11:03:46,129][hulc.datasets.base_dataset][INFO] - loading dataset at /data2/eddie/calvin/task_D_D/training
[2022-08-04 11:03:46,130][hulc.datasets.base_dataset][INFO] - finished loading dataset
[2022-08-04 11:03:46,131][hulc.datasets.disk_dataset][INFO] - Found "ep_start_end_ids.npy" with 31 episodes.
[2022-08-04 11:03:46,215][hulc.datasets.base_dataset][INFO] - loading dataset at /data2/eddie/calvin/task_D_D/validation
[2022-08-04 11:03:46,215][hulc.datasets.base_dataset][INFO] - finished loading dataset
[2022-08-04 11:03:46,216][hulc.datasets.disk_dataset][INFO] - Found "ep_start_end_ids.npy" with 4 episodes.
[2022-08-04 11:03:46,240][hulc.datasets.base_dataset][INFO] - loading dataset at /data2/eddie/calvin/task_D_D/training
[2022-08-04 11:03:46,241][hulc.datasets.base_dataset][INFO] - finished loading dataset
trying to load lang data from:  /data2/eddie/calvin/task_D_D/training/lang_annotations/auto_lang_ann.npy
[2022-08-04 11:03:46,326][hulc.datasets.base_dataset][INFO] - loading dataset at /data2/eddie/calvin/task_D_D/validation
[2022-08-04 11:03:46,326][hulc.datasets.base_dataset][INFO] - finished loading dataset
trying to load lang data from:  /data2/eddie/calvin/task_D_D/validation/lang_annotations/auto_lang_ann.npy
LOCAL_RANK: 0 - CUDA_VISIBLE_DEVICES: [0,1,2,3,4,5,6,7]
Missing logger folder: /home/eddie/git/lad_/runs/2022-08-04/11-03-40_datamodule/datasets=vision_lang/lightning_logs
/home/eddie/.miniconda3/envs/lad/lib/python3.8/site-packages/pytorch_lightning/trainer/data_loading.py:132: UserWarning: The dataloader, train_dataloader, does not have many workers which may be a bottleneck. Consider increasing the value of the `num_workers` argument` (try 64 which is the number of cpus on this machine) in the `DataLoader` init to improve performance.
  rank_zero_warn(
[2022-08-04 11:03:53,919][hulc.wrappers.hulc_wrapper][INFO] - EGL_DEVICE_ID 0 <==> CUDA_DEVICE_ID 0
argv[0]=--width=None
argv[1]=--height=None
[2022-08-04 11:03:54,054][calvin_env.envs.play_table_env][INFO] - Loading EGL plugin (may segfault on misconfigured systems)...
[2022-08-04 11:03:54,055][calvin_env.envs.play_table_env][INFO] - Successfully loaded egl plugin
[2022-08-04 11:03:54,055][calvin_env.envs.play_table_env][INFO] - Connected to server with id: 1
[2022-08-04 11:03:54,057][calvin_env.envs.play_table_env][INFO] - Connected to server with id: 1
[2022-08-04 11:03:54,066][calvin_env.robot.robot][INFO] - Loading robot
[2022-08-04 11:03:54,080][calvin_env.envs.play_table_env][INFO] - Resetting simulation
[2022-08-04 11:03:54,082][calvin_env.envs.play_table_env][INFO] - Setting gravity
[2022-08-04 11:03:54,752][calvin_env.envs.play_table_env][INFO] - Using calvin_env with commit af69b2ac62cdf125297b757ebbdbbed339368ec4.
/home/eddie/.miniconda3/envs/lad/lib/python3.8/site-packages/gym/core.py:329: DeprecationWarning: WARN: Initializing wrapper in old step API which returns one bool instead of two. It is recommended to set `new_step_api=True` to use new step API. This will be the default behaviour in future.
  deprecation(
[2022-08-04 11:03:54,753][hulc.wrappers.hulc_wrapper][INFO] - Initialized PlayTableEnv for device cuda:0
[2022-08-04 11:03:54,757][hulc.wrappers.hulc_wrapper][WARNING] - Environment variable EGL_VISIBLE_DEVICES is already set. Is this intended?
[2022-08-04 11:03:55,881][hulc.wrappers.hulc_wrapper][INFO] - EGL_DEVICE_ID 0 <==> CUDA_DEVICE_ID 0
argv[0]=--width=None
argv[1]=--height=None
[2022-08-04 11:03:56,008][calvin_env.envs.play_table_env][INFO] - Loading EGL plugin (may segfault on misconfigured systems)...
[2022-08-04 11:03:56,009][calvin_env.envs.play_table_env][INFO] - Successfully loaded egl plugin
[2022-08-04 11:03:56,009][calvin_env.envs.play_table_env][INFO] - Connected to server with id: 2
[2022-08-04 11:03:56,011][calvin_env.envs.play_table_env][INFO] - Connected to server with id: 2
[2022-08-04 11:03:56,018][calvin_env.robot.robot][INFO] - Loading robot
[2022-08-04 11:03:56,030][calvin_env.envs.play_table_env][INFO] - Resetting simulation
[2022-08-04 11:03:56,032][calvin_env.envs.play_table_env][INFO] - Setting gravity
[2022-08-04 11:03:56,483][calvin_env.envs.play_table_env][INFO] - Using calvin_env with commit af69b2ac62cdf125297b757ebbdbbed339368ec4.
/home/eddie/.miniconda3/envs/lad/lib/python3.8/site-packages/gym/core.py:329: DeprecationWarning: WARN: Initializing wrapper in old step API which returns one bool instead of two. It is recommended to set `new_step_api=True` to use new step API. This will be the default behaviour in future.
  deprecation(
[2022-08-04 11:03:56,484][hulc.wrappers.hulc_wrapper][INFO] - Initialized PlayTableEnv for device cuda:0
[2022-08-04 11:03:56,888][sentence_transformers.SentenceTransformer][INFO] - Load pretrained SentenceTransformer: paraphrase-MiniLM-L3-v2
[2022-08-04 11:03:57,283][sentence_transformers.SentenceTransformer][INFO] - Use pytorch device: cuda
Batches: 100%|██████████████████████████████████████████| 1/1 [00:00<00:00, 20.74it/s]
Batches: 100%|██████████████████████████████████████████| 1/1 [00:00<00:00, 92.58it/s]
Batches: 100%|██████████████████████████████████████████| 1/1 [00:00<00:00, 86.62it/s]
Batches: 100%|██████████████████████████████████████████| 1/1 [00:00<00:00, 92.09it/s]
[... cut ...]
  File "/home/eddie/.miniconda3/envs/lad/lib/python3.8/site-packages/pytorch_lightning/trainer/callback_hook.py", line 138, in on_train_start
    callback.on_train_start(self, self.lightning_module)
  File "hulc/utils/automatic_lang_annotator_mp.py", line 135, in on_train_start
    outputs: Any,
  File "hulc/utils/automatic_lang_annotator_mp.py", line 125, in init_vars
    self.init_vars(trainer, pl_module)
AttributeError: 'NoneType' object has no attribute 'keys'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
disconnecting id 1 from server
disconnecting id 2 from server
Destroy EGL OpenGL window.
```

mees, 2022-08-05:
Thanks for your report, can you run the command with `datamodule/datasets=vision_only` please? Other parameters that you can set are the window sizes to consider and what percentage of the data should get labeled with language. You might want to check those defaults, in case you want to change them for your application.

lukashermann, 2022-08-05:
You are right, the script was indeed broken. I commited a fix, can you try if it works now? I also added some keys to the config, so now it should be enough to run it as 
`python3 hulc/utils/automatic_lang_annotator_mp.py datamodule.root_data_dir=/data2/eddie/calvin/task_D_D`

ezhang7423, 2022-08-08:
Thank you so much! The new commit works for me.

closed 2022-08-08 by ezhang7423