mees/calvin #25: Feature request: single task datasets

ezhang7423, 2022-08-08:
Hi there! I think it would be really nice if there was a script and dataset for a selection of individual tasks in CALVIN, so that one could test their method on just a single task. I've started working on this already, does it sound like a useful feature?

ezhang7423, 2022-08-08:
If this does sound useful, could I get some advice on what the cleanest/simplest way to implement this in the existing framework is? My current code is a bit hacky...

ezhang7423, 2022-08-08:
I'm talking about training on a single task.

mees, 2022-08-08:
Here you have an example of using task indicators to learn single task rl policies https://github.com/mees/calvin#reinforcement-learning-with-calvin

ezhang7423, 2022-08-08:
Sorry, let be more clear. I specifically mean offline imitation learning through the usage of a subset of the provided dataset filtered on a single task. In essence, breaking up the existing datasets (such as Task D->D) into each individual task that it consists of.

lukashermann, 2022-08-09:
You could try to proceed as follows:
- Use the automatic_lang_annotator to find the episodes that consist of an individual task. You can eps=1 to change the ratio of annotated episodes to 100%. You would have to change these lines in the [`lang_ann.yaml`](https://github.com/mees/calvin/blob/main/calvin_models/conf/lang_ann.yaml#L9) hydra config to two new files in [this](https://github.com/mees/calvin/tree/main/calvin_models/conf/annotations) folder which only contain that task that you are interested in.
- This creates a file `auto_lang_ann.npy` in the lang folder that you specfied in the config. However, since you are not interested in language for a single task, you can create a new `ep_start_end_ids.npy` which you can use to train on a single task: 
```
auto_lang_ann = np.load("auto_lang_ann.npy", allow_pickle=True).item()
ep_start_end_ids = auto_lang_ann["info"]["indx"]
```
- Replace the original ep_start_end_ids in the training and validation folder with the new ep_start_end_ids that only contain those of a single task
- Run a training with vision only by setting `datamodule/datasets=vision_only` 

I haven't tried this approach, so there could be additional steps to make it work :smile:

lukashermann, 2022-09-15:
@ezhang7423 did you get further with what you wanted to do?

ezhang7423, 2022-09-15:
Hi Luka! Thank you so much for your detailed approach. I was able to get further, and will hopefully be able to submit a pull request soon.

closed 2022-09-15 by ezhang7423