mees/calvin #26: Language annotations from the automatic annotation tool

geyuying, 2022-08-11:
Hi,

I visualize the episode with language instructions by sampling 4 images ("rgb_static") within the episode in ABC training set (Task_ABC_D) and some language instructions ("task" and "ann") seem to be wrong as shown below,

[image]

The language instruction of the above episode is "grasp the **blue block** and rotate it right". You can check the example, whose ['info']['indx'] is (744905, 744969). Such cases are not rare in ABC training set, but not in D training and validation set.

I further use the automatic annotation tool to re-annoate the episodes as showed in https://github.com/mees/calvin/issues/24 and get the same task information ("rotate_blue_block_right" in the above example) as  in the downloaded "auto_lang_ann.npy".

Is there anyway to make the language annotations more accurate?

Looking forward to your reply.

Best regards,
Yuying

geyuying, 2022-08-11:
Below is the code that I use for visualization.

```
import numpy as np
import os
import yaml
import matplotlib.pyplot as plt

mdetr_ann_file = 'hulc/dataset/task_ABC_D/training/lang_paraphrase-MiniLM-L3-v2/auto_lang_ann.npy'
mdetr_ann = np.load(mdetr_ann_file, allow_pickle=True).item()
ann_all = mdetr_ann['language']['ann']
task_all = mdetr_ann['language']['task']
indx_all = mdetr_ann['info']['indx']
num_frames = 4
data_dir = 'hulc/dataset/task_ABC_D/training/'
save_dir = 'vis_ABC_lang_training/'

for i, (start_idx, end_idx) in enumerate(indx_all):
    intervals = np.linspace(start=start_idx, stop=end_idx, num=num_frames + 1).astype(int)
    ranges = []
    for index, interv in enumerate(intervals[:-1]):
        ranges.append((interv, intervals[index + 1] - 1))
    frame_idxs = [(x[0] + x[1]) // 2 for x in ranges]
    ann = ann_all[i]

    if 'blue' in ann:
        sub_dir = save_dir + 'blue/'
    elif 'red' in ann:
        sub_dir = save_dir + 'red/'
    elif 'pink' in ann:
        sub_dir = save_dir + 'pink/'
    else:
        continue
    if not os.path.exists(sub_dir):
        os.makedirs(sub_dir)

    save_name = sub_dir + str(i) + '_' + str(start_idx) + '_' + str(end_idx) + '_' + ann + '.jpg'

    for j, file_idx in enumerate(frame_idxs):
        file_path = data_dir + 'episode_' + str(file_idx).zfill(7) + '.npz'
        rgb_static = np.load(file_path)['rgb_static']
        rgb_static = rgb_static / 255
        plt.subplot(2, 2, j + 1)
        plt.imshow(rgb_static)
    plt.savefig(save_name)

    if i>=100:
        break

  ```

lukashermann, 2022-08-15:
I also ran your code and there definitely seems to be a problem. We will investigate what could be the cause of this happening, thank you already for spotting this!

geyuying, 2022-08-15:
Thanks for your reply!

It seems to be caused by the the scene state information. In the above example, I print out the start and end position of the blue block, and rotation does happen.

```
    def rotate_object(
        obj_name, z_degrees, x_y_threshold=30, z_threshold=180, movement_threshold=0.1, start_info=None, end_info=None
    ):
        """
        Returns True if the object with obj_name was rotated more than z_degrees degrees around the z-axis while not
        being rotated more than x_y_threshold degrees around the x or y axis.
        z_degrees is negative for clockwise rotations and positive for counter-clockwise rotations.
        """
        obj_start_info = start_info["scene_info"]["movable_objects"][obj_name]
        obj_end_info = end_info["scene_info"]["movable_objects"][obj_name]
        start_orn = R.from_quat(obj_start_info["current_orn"])
        end_orn = R.from_quat(obj_end_info["current_orn"])
        rotation = end_orn * start_orn.inv()
        x, y, z = rotation.as_euler("xyz", degrees=True)

        start_pos = np.array(obj_start_info["current_pos"])
        end_pos = np.array(obj_end_info["current_pos"])
```

lukashermann, 2022-09-16:
Thanks again for finding this bug!
We describe the fix for the problem [here](https://github.com/mees/calvin#16-sep-2022).

closed 2022-09-16 by lukashermann