mees/calvin #32: Major concern about evaluation

ezhang7423, 2022-12-18:
Hi there!
I've found that rolling out ground truth trajectories (labelled by the language annotator) from the dataset is not always evaluated to be successful by the Tasks.get_task_info. This seems to be quite concerning. Perhaps I've done something wrong on my end?
[image]

To reproduce, I have forked the repo with minimal changes here: https://github.com/ezhang7423/calvin.
The only difference is in line 47 in `calvin_modesl/calvin_agent/evaluation/evaluate_policy_singlestep.py`, where instead of rolling out the model I roll out the dataset actions.

The exact commands I ran from beginning to end: 
```
# set up environment
git clone git@github.com:ezhang7423/calvin.git --recursive
cd calvin
conda create --name calvin python=3.8
conda activate calvin
pip install setuptools==57.5.0 torchmetrics==0.6.0
./install.sh

# get pretrained weights and fix the config.yaml
cp ./D_D_static_rgb_baseline/.hydra/config.yaml ./tmp.yaml
wget http://calvin.cs.uni-freiburg.de/model_weights/D_D_static_rgb_baseline.zip
unzip D_D_static_rgb_baseline.zip
unzip D_D_static_rgb_baseline.zip
mv ./tmp.yaml ./D_D_static_rgb_baseline/.hydra/config.yaml

# get data
cd dataset
./download_data.sh D
cd ../

# run the evaluation
python calvin_models/calvin_agent/evaluation/evaluate_policy_singlestep.py --dataset_path $DATA_GRAND_CENTRAL/task_D_D/ --train_folder ./D_D_static_rgb_baseline/ --checkpoint D_D_static_rgb_baseline/mcil_baseline.ckpt
```

notmahi, 2022-12-20:
Assuming that there is some noise in the initialization of the robot in the simulator, _should_ we expect open-loop rollout from the dataset to be a success? I have a hunch that the answer is no. In fact, if open-loop rollouts get good success, why should anyone bother learning a policy?

ezhang7423, 2022-12-20:
Turn off led gets 0/10 every time, so I'm almost sure that something is broken. 

However, this is a good point. I will try rerunning after resetting the environment to the inital state. After this, we should expect that every rollout should be successful, as the environment is deterministic. Outside of some floating point error, every state in the trajectory should match the online rollout.

lukashermann, 2022-12-22:
Thank you for this comprehensive issue. I did some investigations and there are multiple things going on. 

The first thing I would like to comment on is that these are NOT the ground truth trajectories labeled by the language annotator. This evaluation mimics what happens in the [rollout callback](https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/rollout/rollout.py). It is actually not very informative and we left it in the repository more for legacy reasons. The rollout callback goes through the validation sequences and checks if a task was achieved in that sequence until you have up to 10 instances per task (this is similar to how we label language instructions, but there are some differences in detail). The sequence indices are saved in `task_to_id_dict` and that's what you also load [here](https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/evaluation/evaluate_policy_singlestep.py#L20) in the evaluation.  Since validation sequences are sampled in order with a sliding window, it's likely that some of those 10 sequences are from the same original demonstration of the task, just shifted by some steps (you can also see this if you look at the indices in the `task_to_id_dict`. This is of course not optimal, but we didn't really use this kind of rollout and evaluation much after we started using the proper long-horizon evaluation from a neutral starting position (although it still serves as a good sanity check as we see here).

So here are some reasons for the problems:

- There was indeed a bug in the `turn_off_led` task in `calvin_env`, I will commit a fix, thank you! The good thing is that this bug does not affect the dataset.
- The open loop execution of 32 actions is not 100% identical to resetting to the final state. Sometimes the difference between success and fail can be a tiny bit of movement. I looked at the failure cases and saw things happening like:
-- the block getting grasped on the very edge, in the open loop execution it slips out of the fingers.
-- the robot moving the switch by only the slightest touch to the handle (maybe even friction forces), in the open loop it misses the handle
--  many tasks require a movement of an object or door by a certain distance (e.g. pushing requires a block to move 10 cm). It's in the nature of how the indices in `task_to_id_dict` are created, that they often reside in extreme cases. Compounding small errors of the open loop execution can have the effect, that for example a block that is originally pushed 10 cm, gets only pushed 9.9 cm.
- In January and February we pushed some small changes to the task definition of the rotating and pushing task (mostly to get rid of edge cases and false positive detections), but the model you used for evaluation was trained before, so the indices in `task_to_id_dict` are partially wrong due to the redefinition of the tasks. This is by itself not a very big problem, and it doesn't affect the long-horizon evaluation. However, the language annotations were also created before these changes, so some sequences labeled as a rotating or pushing task might not qualify as that under the new definition. I think the practical implications are not so big of this problem but we will discuss if we should rerun the labeling. 
- Some indices in the `task_to_id_dict` actually contain the wrong task (you can see this when you run the script with `--debug`. This is weird and I don't know why it happened. I reran the first 2 epochs to get a new and up-to-date `task_to_id_dict` and it didn't happen again, so it's most likely due to some changes that happened since we trained the model.

ezhang7423, 2022-12-27:
Hi Luka, thank you so much for the detailed reply. I think the explanation regarding the small error accumulation being the chief cause for the failure makes sense. I hope that the indices in the task_to_id_dict can be updated to the correct labels to make things a bit easier for us. I see that the `turn_off_led` has been fixed with this commit, thank you! https://github.com/mees/calvin_env/commit/797142c588c21e76717268b7b430958dbd13bf48

closed 2022-12-27 by ezhang7423

lukashermann, 2022-12-27:
We're planning to relabel the language instructions and retrain the models, it should be ready in 1 - 2 weeks!