mees/calvin #36: Update Rollouts Checkpointing to new PL version

mees, 2023-01-02:
Update Rollout callback [0] in new lightning branch to use new API [1] for adding state to callback checkpoints. Check that the short horizon rollouts still work in the new branch.
[0] https://github.com/mees/calvin/blob/new_lightning/calvin_models/calvin_agent/rollout/rollout.py#L400

[1] https://pytorch-lightning.readthedocs.io/en/stable/extensions/callbacks_state.html

lukashermann, 2023-01-02:
but is there any advantage over using `on_load_checkpoint`? https://pytorch-lightning.readthedocs.io/en/stable/extensions/callbacks.html#on-load-checkpoint

mees, 2023-01-03:
I thought `on_load_checkpoint` had been deprecated, but it seems to still work, my bad. I have enabled the short horizon rollouts and added a missing dependency in the new branch.

closed 2023-01-03 by mees

lukashermann, 2023-01-03:
Ah yes, I disabled the short horizon rollouts by default because I thought they are not so useful and take a lot of time. But we can leave it as it was originally.