mees/calvin #39: Error in automatic annotation

jsw7460, 2023-02-01:
Thank you for supporting great benchmark.

I'm trying to generate language annotation for unlabeled steps(frames). 

I used [automatic_lang_annotator_mp.py](https://github.com/mees/calvin/blob/32b4a355397fb594f5aac34506749f352be71e46/calvin_models/calvin_agent/utils/automatic_lang_annotator_mp.py)
and adjust the eps Hyperparameter [here](https://github.com/mees/calvin/blob/main/calvin_models/conf/lang_ann.yaml#L21) to be 0.5. 

I encountered IndexError:

[image]

(Running the same code with eps=0.01 worked fine for me.)

I have two questions
1. Should I decrease the epsilon to fix the error?
2. How big should epsilon be to have no unlabeled data?

lukashermann, 2023-02-12:
@jsw7460 sorry for addressing this so late. I just reproduced the error, will get back to you when I find out what causes it. Did you investigate further in the meantime?

lukashermann, 2023-02-12:
To your questions, yes a lower eps will prevent it from happening. I successfully annotated it with `eps=0.1`
Note that not every sampled window can be labeled with a language instruction (because the prerequisite is that a task is detected). I don't know what's the minimum epsilon for which all possible labels are created, but usually setting `eps=1` should work (`eps=0.5` of course too)