mees/calvin #40: some inconsistencies in the dataset

denizyuret, 2023-02-19:
My student @emrecanacikgoz and I are experimenting with the calvin dataset and noticed a couple of minor inconsistencies that we wanted to bring to your attention (in case somebody relies on exact indices for alignment etc):

* scene_info.npy for D/training indicates calvin_scene_A but it should be calvin_scene_D.
* ABCD/training includes several frames from scene_D that are also in the validation set (seems like an off by one error): 37682, 53818, 244284,420498.

lukashermann, 2023-02-19:
When did you download the dataset? We fixed the problem in the scene_info.npy in September last year, see [https://github.com/mees/calvin#16-sep-2022](https://github.com/mees/calvin#16-sep-2022), unless something went wrong and the dataset didn't get updated on the server.

lukashermann, 2023-02-19:
Regarding your second concern, thanks for spotting this! However, these duplicate frames are not really a problem, since they are not present in the `ep_start_end_ids.npy` of the training set. We should still remove them at one point.

denizyuret, 2023-02-19:
> When did you download the dataset? We fixed the problem in the scene_info.npy in September last year, see https://github.com/mees/calvin#16-sep-2022, unless something went wrong and the dataset didn't get updated on the server.

Strangely the debug directory is dated May, others November. Maybe our IT dept did something wrong downloading. @emrecanacikgoz what instructions did you give them for the download?

denizyuret, 2023-02-19:
> Regarding your second concern, thanks for spotting this! However, these duplicate frames are not really a problem, since they are not present in the `ep_start_end_ids.npy` of the training set. We should still remove them at one point.

Thanks! I am just getting familiar with the dataset: ep_start_end_ids.npy indicate the start-end of a single play session (in which there should not be any discontinuities) correct?

emrecanacikgoz, 2023-02-19:
> 

I asked our IT to follow the instructions from the repository by giving the corresponding [link](https://github.com/mees/calvin/tree/main/dataset), on the 21st of November 2022.

lukashermann, 2023-02-19:
> Thanks! I am just getting familiar with the dataset: ep_start_end_ids.npy indicate the start-end of a single play session (in which there should not be any discontinuities) correct?

Exactly!

I am investigating the error in the dataset scene_info.npy

lukashermann, 2023-02-23:
You are right, it is a wrong scene annotation in the dataset. For now this doesn't affect much unless you run the language annotation script again, but it has to be fixed. I will update the dataset on the server. Thank you for spotting this error!

denizyuret, 2023-02-23:
Thanks @lukashermann. I had a few other questions maybe you can help with (these are not strictly related to dataset inconsistencies, I can open another issue or send an email if you think is more appropriate):

1. actions: consist of xyz, euler angles, and gripper state. Do we really control the robot by sending where we want the gripper to go 1/30th of a second from now?  (I am not a roboticist, so apologies if this is a dumb question).

2. the actions and robot_obs refer to "tcp" (tool center point), or "EE" (end effector?) coordinates. Where exactly is this point? Is it in the middle of the two fingertips?

3. the rel_actions are defined as "Relative cartesian displacement wrt the gripper frame" as opposed to actions "Absolute cartesian pose wrt the world frame". So is rel_actions the difference between the next frame we want and the current frame? Isn't the tcp/ee at the origin of the "gripper frame" in which case it should never move? How do we go between gripper frame and world frame, or specifically how can I derive rel_actions from the other data given?

4. the controllers (button, door, switch, drawer) are each described by a single number in the data indicating their one degree of freedom. However their absolute position in world frame is not given. How can we obtain their positions from the URDF files, YAML files or any other files in calvin_env describing the scenes?

lukashermann, 2023-02-23:
1. We control the gripper with binary actions (open / close), but you are right, the control frequency is 30 Hz, so you can send a different action every frame, which makes little sense of course. In practice we never had a problem with the output of the network for gripper action, i.e. we didn't observe jittering of the gripper. Also note that you are not bound to controlling the robot with 30 Hz, you can e.g. do action repeats and effectively use only 15 Hz (or less). 
2. TCP and end effector are often used interchangeably and indeed, we define the TCP to be in the middle of the fingertips. The exact position is defined in this [URDF file](https://github.com/mees/calvin_env/blob/main/data/franka_panda/panda_longer_finger.urdf#L369).
3. We used [this](https://github.com/mees/calvin_env/blob/main/calvin_env/utils/utils.py#L160) function to convert absolute actions to relative actions at the time of the data collection. The description "Relative cartesian displacement wrt the gripper frame" is unfortunately badly chosen, sorry for the confusion. For two frames at t1 and t2 you would define the relative action as the relative movement of the tcp pose in world position at t1 and t2 (for x,y,z that's just a displacement vector and for the orientation you get relative euler angles). The way of defining the rotation component of the relative action is hacky, but it works in practice.  So the relative actions are not really in gripper frame, but they refer to a relative movement of the gripper in world frame. You can however transform them into the gripper frame, which is what we are doing in the follow up paper hulc see [here](https://github.com/lukashermann/hulc/blob/main/hulc/models/decoders/utils/gripper_control.py) and [here](https://github.com/lukashermann/hulc/blob/main/hulc/models/decoders/deterministic_decoder.py#L74). _- Isn't the tcp/ee at the origin of the "gripper frame" in which case it should never move?_ You can represent the tcp of t2 in the frame of tcp at t1. _- How do we go between gripper frame and world frame, or specifically how can I derive rel_actions from the other data given?_ You have the tcp position and orientation for every frame. From this you can compute anything you want, for example you can construct a 4x4 homogeneous matrix and compute the transformations between two frames (either in tcp or world frame). Don't hesitate further questions to clarify this, I know it can be quite confusing.
4. If you want the positions you would have to call `pybullet.getLinkState` in calvin_env [as defined here](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.3v8gjd1epcrt). However, the position of the link is not necessarily at the handle, so if you want to have the position of the handles at every step, you can create a dummy link in the urdf files (similar to the tcp for the robot).

denizyuret, 2023-02-23:
Thank you! This helps a lot.

commit 4b35b135 (2023-02-24): Update scene info of D_D to fix #40
closed 2023-02-24 by mees