mees/calvin #43: ModuleNotFoundError: No module named 'calvin_agent.datasets.play_data_module'

OOXXXXOO, 2023-03-20:
```bash
# ----- Load ckpt from  /home/pi/workspace/tanwenxuan/Project/agi-perception-roboton/resources/model/hulc-mtlc/D_D_static_rgb_baseline/mcil_baseline.ckpt
Iter with ckpt: /home/pi/workspace/tanwenxuan/Project/agi-perception-roboton/resources/model/hulc-mtlc/D_D_static_rgb_baseline/mcil_baseline.ckpt
Traceback (most recent call last):
  File "/home/pi/anaconda3/envs/hulc_venv/lib/python3.8/site-packages/hydra/_internal/utils.py", line 585, in _locate
    import_module(mod)
  File "/home/pi/anaconda3/envs/hulc_venv/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
```

```bash

Traceback (most recent call last):
  File "calvin_agent/evaluation/evaluate_policy.py", line 252, in <module>
    main()
  File "calvin_agent/evaluation/evaluate_policy.py", line 241, in main
    model, env, _ = get_default_model_and_env(
  File "/home/pi/workspace/tanwenxuan/Project/agi-perception-roboton/third_party/calvin/calvin_models/calvin_agent/evaluation/utils.py", line 35, in get_default_model_and_env
    data_module = hydra.utils.instantiate(cfg.datamodule, num_workers=0)
  File "/home/pi/anaconda3/envs/hulc_venv/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 180, in instantiate
    return instantiate_node(config, *args, recursive=_recursive_, convert=_convert_)
  File "/home/pi/anaconda3/envs/hulc_venv/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 240, in instantiate_node
    _target_ = _resolve_target(node.get(_Keys.TARGET))
  File "/home/pi/anaconda3/envs/hulc_venv/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 104, in _resolve_target
    return _locate(target)
  File "/home/pi/anaconda3/envs/hulc_venv/lib/python3.8/site-packages/hydra/_internal/utils.py", line 587, in _locate
    raise ImportError(
ImportError: Encountered error: `No module named 'calvin_agent.datasets.play_data_module'` when loading module 'calvin_agent.datasets.play_data_module.PlayDataModule'

```

mees, 2023-04-12:
Hi, is calvin listed as installed when you do `pip list` ?
Can you try calling the code from inside the calvin_agent folder as shown in the readme?
```
$ cd $CALVIN_ROOT/calvin_models/calvin_agent
$ python evaluation/evaluate_policy.py --dataset_path <PATH/TO/DATASET> --train_folder <PATH/TO/TRAINING/FOLDER>
```

emrecanacikgoz, 2023-05-23:
Same issue, I followed the readme. Any updates?

Stardust-y, 2023-06-06:
same errors, I think it's because there's no play_data_module.py released under  /calvin_models/calvin_agent folder

tarekdawey, 2023-11-01:
Same error. It can't find play_data_module in the directory  /calvin_models/calvin_agent

ModuleNotFoundError: No module named 'calvin_agent.datasets.play_data_module'

mees, 2023-12-07:
The config has been updated and soon work now. Repen if the issue still persists.

closed 2023-12-07 by mees