mees/calvin #53: TypeError: Error instantiating 'calvin_agent.utils.transforms.NormalizeVector' : new(): data must be a sequence (got float)

gauraviiita, 2023-08-08:
Hello Oier, thank you for publishing your work. I am interested to work in this area. I am trying to implement your repository on my GPU desktop. I am getting an error when running the command
$ python training.py datamodule.root_data_dir=/media/robita/second_ssd/Gaurav/calvin/dataset/ datamodule/datasets=vision_lang_shm

Please let me know what is the mistake. What should I do to remove this error?

15 Extracting /media/robita/second_ssd/Gaurav/calvin/dataset/training/50steps.tar.xz to /media/robita/second_ssd/Gaurav/calvin/dataset/training
16 Downloading http://www2.informatik.uni-freiburg.de/~meeso/50steps.tar.xz to /media/robita/second_ssd/Gaurav/calvin/dataset/validation/50steps.tar.xz
17 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 12960424/12960424 [00:01<00:00, 11767561.95it/s]
18 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 14339.50it/s]
19 Extracting /media/robita/second_ssd/Gaurav/calvin/dataset/validation/50steps.tar.xz to /media/robita/second_ssd/Gaurav/calvin/dataset/validation
20 [2023-08-09 04:11:04,627][calvin_agent.datasets.utils.shared_memory_utils][INFO] - Loading train language episodes into shared memory. (progress bar shows only worker process 0).
21 [2023-08-09 04:11:04,813][calvin_agent.datasets.utils.shared_memory_utils][INFO] - Loading val language episodes into shared memory. (progress bar shows only worker process 0).
22 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 7681.88it/s]
23 Error executing job with overrides: ['datamodule.root_data_dir=/media/robita/second_ssd/Gaurav/calvin/dataset/', 'datamodule/datasets=vision_lang_shm']
24 Traceback (most recent call last):
25   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 62, in _call_target
26     return _target_(*args, **kwargs)
27   File "/media/robita/second_ssd/Gaurav/calvin/calvin_models/calvin_agent/utils/transforms.py", line 22, in __init__
28     self.std = torch.Tensor(std)
29 TypeError: new(): data must be a sequence (got float)
30 During handling of the above exception, another exception occurred:
31 Traceback (most recent call last):
32   File "training.py", line 68, in train
33     trainer.fit(model, datamodule=datamodule, ckpt_path=chk)  # type: ignore
34   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 603, in fit
35     call._call_and_handle_interrupt(
36   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/call.py", line 38, in _call_and_handle_interrupt
37     return trainer_fn(*args, **kwargs)
38   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 645, in _fit_impl
39     self._run(model, ckpt_path=self.ckpt_path)
40   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1037, in _run
41     self._call_setup_hook()  # allow user to setup lightning_module in accelerator environment
42   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1284, in _call_setup_hook
43     self._call_lightning_datamodule_hook("setup", stage=fn)
44   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1361, in _call_lightning_datamodule_hook
45     return fn(*args, **kwargs)
46   File "/media/robita/second_ssd/Gaurav/calvin/calvin_models/calvin_agent/datasets/calvin_data_module.py", line 79, in setup
47     self.train_transforms = {
48   File "/media/robita/second_ssd/Gaurav/calvin/calvin_models/calvin_agent/datasets/calvin_data_module.py", line 80, in <dictcomp>
49     cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train
50   File "/media/robita/second_ssd/Gaurav/calvin/calvin_models/calvin_agent/datasets/calvin_data_module.py", line 80, in <listcomp>
51     cam: [hydra.utils.instantiate(transform) for transform in transforms.train[cam]] for cam in transforms.train
52   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 180, in instantiate
53     return instantiate_node(config, *args, recursive=_recursive_, convert=_convert_)
54   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 249, in instantiate_node
55     return _call_target(_target_, *args, **kwargs)
56   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 64, in _call_target
57     raise type(e)(
58   File "/home/robita/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 62, in _call_target
59     return _target_(*args, **kwargs)
60   File "/media/robita/second_ssd/Gaurav/calvin/calvin_models/calvin_agent/utils/transforms.py", line 22, in __init__
61     self.std = torch.Tensor(std)
62 TypeError: Error instantiating 'calvin_agent.utils.transforms.NormalizeVector' : new(): data must be a sequence (got float)
63 Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

mees, 2023-09-06:
Hi there,
first download one of the datasets and then on the command line argument `datamodule.root_data_dir=` you need to specify the folder with chosen split.

gauraviiita, 2023-09-09:
Thank you for getting back to me.

I am trying to run using a debug dataset.
I am using the command as you said
$ python training.py datamodule.root_data_dir=/home/gaurav/Desktop/calvin/dataset/calvin_debug_dataset/ datamodule/datasets=vision_lang_shm

And it works.

But now I am getting another error. Please let me know how can I remove this error and run the entire code successfully.

The error is as follows:

Epoch 0:   0%|                                                                                                                                                                     | 0/139 [00:00<?, ?it/s][2023-09-09 21:45:43,525][calvin_agent.models.mcil][INFO] - Start training epoch 0
Epoch 0:   3%|███▊                                                                                                                                  | 4/139 [00:09<05:14,  2.33s/it, loss=41.9, v_num=5-10]Error executing job with overrides: ['datamodule.root_data_dir=/home/gaurav/Desktop/calvin/dataset/calvin_debug_dataset', 'datamodule/datasets=vision_lang_shm']
Traceback (most recent call last):
  File "training.py", line 68, in train
    trainer.fit(model, datamodule=datamodule, ckpt_path=chk)  # type: ignore
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 603, in fit
    call._call_and_handle_interrupt(
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/call.py", line 38, in _call_and_handle_interrupt
    return trainer_fn(*args, **kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 645, in _fit_impl
    self._run(model, ckpt_path=self.ckpt_path)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1098, in _run
    results = self._run_stage()
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1177, in _run_stage
    self._run_train()
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1200, in _run_train
    self.fit_loop.run()
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/loop.py", line 199, in run
    self.advance(*args, **kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/fit_loop.py", line 267, in advance
    self._outputs = self.epoch_loop.run(self._data_fetcher)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/loop.py", line 199, in run
    self.advance(*args, **kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/epoch/training_epoch_loop.py", line 214, in advance
    batch_output = self.batch_loop.run(kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/loop.py", line 199, in run
    self.advance(*args, **kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/batch/training_batch_loop.py", line 88, in advance
    outputs = self.optimizer_loop.run(optimizers, kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/loop.py", line 199, in run
    self.advance(*args, **kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/optimization/optimizer_loop.py", line 200, in advance
    result = self._run_optimization(kwargs, self._optimizers[self.optim_progress.optimizer_position])
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/optimization/optimizer_loop.py", line 247, in _run_optimization
    self._optimizer_step(optimizer, opt_idx, kwargs.get("batch_idx", 0), closure)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/optimization/optimizer_loop.py", line 357, in _optimizer_step
    self.trainer._call_lightning_module_hook(
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/trainer/trainer.py", line 1342, in _call_lightning_module_hook
    output = fn(*args, **kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/core/module.py", line 1661, in optimizer_step
    optimizer.step(closure=optimizer_closure)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/core/optimizer.py", line 169, in step
    step_output = self._strategy.optimizer_step(self._optimizer, self._optimizer_idx, closure, **kwargs)
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/strategies/strategy.py", line 234, in optimizer_step
    return self.precision_plugin.optimizer_step(
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/plugins/precision/native_amp.py", line 85, in optimizer_step
    closure_result = closure()
  File "/home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/site-packages/pytorch_lightning/loops/optimization/optimizer_loop.py", line 147, in __call__
    self._result = self.closure(*args, **kwargs)
[... cut ...]
Epoch 0:   3%|███▊                                                                                                                                  | 4/139 [00:27<15:39,  6.96s/it, loss=41.9, v_num=5-10]
(calvin_venv) gaurav@gaurav-GF65-Thin-10UE:~/Desktop/calvin/calvin_models/calvin_agent$ /home/gaurav/anaconda3/envs/calvin_venv/lib/python3.8/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 8 leaked shared_memory objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '

Thank you for your guidance

mees, 2023-09-09:
try reducing the batch size, as it says you are out of memory. Also I am not sure if the shm loader works with the debug dataset, have you tried without? Ideally try to run one of the real dataset splits.

gauraviiita, 2023-09-10:
Yes, It is working now for the debug dataset also. But yes, now I will also use the D dataset for the training.
Thanks a lot for your support.

closed 2023-09-10 by mees