mees/calvin #55: How to render in a high-resolution mode?

Ericonaldo, 2023-10-12:
Hi I am using CALVIN for my research and I want to show some visualizations in high resolution (200x200 by default). I tried to change the camera size but it won't work. I think there must be solutions as I noticed that the visualization demo shown on the home page is good.

2488583886, 2023-11-27:
My email address is [ybfuyujie@gmail.com](mailto:ybfuyujie@gmail.com), thanks so much!

mees, 2023-11-27:
> Hi I am using CALVIN for my research and I want to show some visualizations in high resolution (200x200 by default). I tried to change the camera size but it won't work. I think there must be solutions as I noticed that the visualization demo shown on the home page is good.

Good question, you can check this cript that we used in the past https://github.com/mees/calvin_env/blob/main/calvin_env/scripts/record_video_icra.py

closed 2023-11-27 by mees