mees/calvin #57: ModuleNotFoundError: No module named 'calvin_agent.datasets.play_data_module'

tarekdawey, 2023-11-01:
Hi,

I am trying to run this:  python evaluation/evaluate_policy.py --dataset_path /home/systemtec/calvin/dataset/task_D_D --train_folder /home/systemtec/calvin/D_D_static_rgb_baseline --checkpoint /home/systemtec/calvin/D_D_static_rgb_baseline/mcil_baseline.ckpt --debug

ModuleNotFoundError: No module named 'calvin_agent.datasets.play_data_module'

tarekdawey, 2023-11-01:
Below is the pip list:

Package                       Version      Editable project location
----------------------------- ------------ ---------------------------------------
actionlib                     1.14.0
aiohttp                       3.8.6
aiosignal                     1.3.1
angles                        1.9.13
antlr4-python3-runtime        4.8
appdirs                       1.4.4
async-timeout                 4.0.3
attrs                         23.1.0
black                         23.10.0
bondpy                        1.8.6
Calvin                        0.0.1        /home/systemtec/calvin/calvin_models
calvin-env                    0.0.1        /home/systemtec/calvin/calvin_env
camera-calibration            1.17.0
camera-calibration-parsers    1.12.0
catkin                        0.8.10
certifi                       2023.7.22
cffi                          1.16.0
cfgv                          3.4.0
charset-normalizer            3.3.0
click                         8.1.7
cloudpickle                   3.0.0
cmake                         3.18.4
colorlog                      6.7.0
contourpy                     1.1.1
controller-manager            0.19.6
controller-manager-msgs       0.19.6
coverage                      7.3.2
cv-bridge                     1.16.2
cycler                        0.12.1
decorator                     4.4.2
diagnostic-analysis           1.11.0
diagnostic-common-diagnostics 1.11.0
diagnostic-updater            1.11.0
distlib                       0.3.7
docker-pycreds                0.4.0
dynamic-reconfigure           1.7.3
exceptiongroup                1.1.3
filelock                      3.12.4
flake8                        6.1.0
fonttools                     4.43.1
freetype-py                   2.4.0
frozenlist                    1.4.0
fsspec                        2023.9.2
gazebo_plugins                2.9.2
gazebo_ros                    2.9.2
gencpp                        0.7.0
geneus                        3.0.0
genlisp                       0.4.18
genmsg                        0.6.0
gennodejs                     2.0.2
genpy                         0.6.15
gitdb                         4.0.10
GitPython                     3.1.38
gym                           0.26.2
gym-notices                   0.0.8
huggingface-hub               0.17.3
hydra-colorlog                1.2.0
hydra-core                    1.1.1
identify                      2.5.30
idna                          3.4
image-geometry                1.16.2
imageio                       2.31.5
imageio-ffmpeg                0.4.9
importlib-metadata            6.8.0
importlib-resources           6.1.0
iniconfig                     2.0.0
[... 151 lines cut ...]
tf2-geometry-msgs             0.7.7
tf2-kdl                       0.7.7
tf2-py                        0.7.7
tf2-ros                       0.7.7
threadpoolctl                 3.2.0
tokenizers                    0.14.1
tomli                         2.0.1
topic-tools                   1.16.0
torch                         1.13.1
torchmetrics                  1.2.0
torchvision                   0.14.1
tqdm                          4.66.1
transformers                  4.34.0
trimesh                       4.0.0
typing_extensions             4.8.0
tzdata                        2023.3
urdfdom-py                    0.4.6
urdfpy                        0.0.22
urllib3                       2.0.7
virtualenv                    20.24.5
wandb                         0.15.12
wheel                         0.41.2
xacro                         1.14.16
yarl                          1.9.2
zipp                          3.17.0

RamyE, 2023-12-05:
I have had to personally make some modifications to the config file and to the evaluate_policy_singlestep script to get it to run

RamyE, 2023-12-05:
config.yaml after updates
```
callbacks:
  rollout:
    tasks:
      _target_: calvin_env.envs.tasks.Tasks
      tasks:
        rotate_red_block_right:
        - rotate_object
        - block_red
        - -60
        rotate_red_block_left:
        - rotate_object
        - block_red
        - 60
        rotate_blue_block_right:
        - rotate_object
        - block_blue
        - -60
        rotate_blue_block_left:
        - rotate_object
        - block_blue
        - 60
        rotate_pink_block_right:
        - rotate_object
        - block_pink
        - -60
        rotate_pink_block_left:
        - rotate_object
        - block_pink
        - 60
        push_red_block_right:
        - push_object
        - block_red
        - 0.1
        - 0
        push_red_block_left:
        - push_object
        - block_red
        - -0.1
        - 0
        push_blue_block_right:
        - push_object
        - block_blue
        - 0.1
        - 0
        push_blue_block_left:
        - push_object
        - block_blue
        - -0.1
        - 0
        push_pink_block_right:
        - push_object
        - block_pink
        - 0.1
        - 0
        push_pink_block_left:
        - push_object
        - block_pink
        - -0.1
        - 0
        move_slider_left:
        - move_door_rel
        - base__slide
        - 0.15
        move_slider_right:
        - move_door_rel
        - base__slide
        - -0.15
        open_drawer:
[... 417 lines cut ...]
  _target_: calvin_agent.models.mcil.MCIL
  _recursive_: false
  kl_beta: ${loss.kl_beta}
loss:
  kl_beta: 0.001
training:
  lr: 0.0001
trainer:
  gpus: 8
  precision: 16
  val_check_interval: 1.0
  max_epochs: 100
logger:
  _target_: pytorch_lightning.loggers.WandbLogger
  save_dir: .
  name: play_lmp
  group: play_lmp
  log_model: false
  project: multi_play
  entity: multimodal_control
  id: ???
seed: 42
log_dir: ../
slurm: true
```

RamyE, 2023-12-05:
I also made this change to evaluate_policy_singlestep
[image]

lukashermann, 2023-12-07:
Yes, this is unfortunately due to the config file of the MCIL baseline not being updated after our last major refactoring. I will try to replace it as soon as possible.