mees/calvin #58: Request for Guidance on Secondary Development Based on CALIVIN Benchmark

Kira0096, 2023-11-06:
Hello,

I am currently working on a project that involves the use of the CALIVIN benchmark. I am interested in performing secondary development on this dataset to generate more tasks. However, I am not quite sure about the best approach to take.

Could you kindly provide some documentation or examples that could guide me through the process? Any assistance on how to manipulate the dataset to generate more tasks would be greatly appreciated.

Thank you in advance for your help.

2488583886, 2023-11-27:
My email address is [ybfuyujie@gmail.com](mailto:ybfuyujie@gmail.com), thanks so much!

lukashermann, 2024-01-03:
@Kira0096 did you mean recording new play data with manual teleoperation or just defining and detecting new tasks in the existing environment interactions?

houyaokun, 2024-01-11:
> @Kira0096 did you mean recording new play data with manual teleoperation or just defining and detecting new tasks in the existing environment interactions?

If I want to generate a small-scale dataset in my own robotic arm simulation environment, mimicking the format of the Calvin dataset, what preparations should I make? Or where should I refer to? Thanks a lot. :)