mees/calvin #61: [Generate More Data] Can I use the simulator to generate more data?

Aopolin-Lv, 2023-11-27:
Hi! I would like to generate some more data of the existing task, such as `push_red_block_left`. Can I build the relevant data through the repo?

COST-97, 2024-02-22:
@mees Hello:
Could you provide a demo of an expert policy (e.g. a policy obtained using the RoboFlamingo method) to produce more data? under the same Env A, B, C, and D, under the same language annotations?
Thank you!

lukashermann, 2024-02-26:
According to the RoboFlamingo [Github repo](https://github.com/RoboFlamingo/RoboFlamingo), they uploaded pre-trained models on [hugging face ](https://huggingface.co/roboflamingo/RoboFlamingo/blob/main/README.md).

filipul1s, 2025-02-13:
> [@mees](https://github.com/mees) Hello: Could you provide a demo of an expert policy (e.g. a policy obtained using the RoboFlamingo method) to produce more data? under the same Env A, B, C, and D, under the same language annotations? Thank you!
Do you now know how to make a dataset in CALVIN format? I would like to ask some advice

filipul1s, 2025-02-13:
> [@mees](https://github.com/mees) Hello: Could you provide a demo of an expert policy (e.g. a policy obtained using the RoboFlamingo method) to produce more data? under the same Env A, B, C, and D, under the same language annotations? Thank you!

**Do you now know how to make a dataset in CALVIN format? I would like to ask some advice.**