mees/calvin #62: Dataset downloading issue

SiyuanHuang95, 2023-11-29:
Thanks for your sharing!

1. Finding the wget is too slow, also since that large file is compressed into one file, not robust for the internet temporary issue
2. Possible to hold this dataset with multiple subfiles? And with the Google Drive?
3. Sometimes it rasies with: ERROR 503: Service Unavailable.
4. Would like to know, can we also obtain the more high-quality / high resolution images for this dataset?

mees, 2023-12-29:
Thanks for pointing this out, I agree that it's not the best way to serve the dataset right now. I will try to look into some alternatives.

SiyuanHuang95, 2023-12-29:
Thanks for your reply. And how about the HQ images issue, was that possible?

mees, 2023-12-30:
I don't think so, one would need to re-render the whole splits and host them separately.

SiyuanHuang95, 2023-12-31:
Okay, or can you provide some script for the HQ images rendering?

SiyuanHuang95, 2024-01-03:
Hi @lukashermann,

Great thanks for your information. With the current information, I think I am able to render the desired image resolution on my side. Thanks again.

closed 2024-01-03 by SiyuanHuang95