mees/calvin #65: error in install.sh - pyhash package

mnm117, 2024-01-05:
Hello, when I try to " sh install.sh" I faced below error:
```

Collecting pyhash (from Calvin==0.0.1)
  Using cached pyhash-0.9.3.tar.gz (602 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [11 lines of output]
      /home/mohsen/anaconda3/envs/RoboFlamingo/lib/python3.8/site-packages/setuptools/__init__.py:80: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
      !!
      
              ********************************************************************************
              Requirements should be satisfied by a PEP 517 installer.
              If you are using pip, you can try `pip install --use-pep517`.
              ********************************************************************************
      
      !!
        dist.fetch_build_eggs(dist.setup_requires)
      error in pyhash setup command: use_2to3 is invalid.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```
the I tried " pip install --use-pep517 pyhash " and the error is exist. Can anyone help ?

`ollecting pyhash
  Using cached pyhash-0.9.3.tar.gz (602 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in pyhash setup command: use_2to3 is invalid.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.`

lukashermann, 2024-01-05:
Hi @mnm117, you have to downgrade `setuptools` to a version below 58 before installing `pyhash`.

qureshinomaan, 2024-01-06:
The version that worked for me : setuptools~=57.5.0

qureshinomaan, 2024-01-06:
Hi!
I was able to resolve this error using a different version of setuptools. However, I am stuck while trying to train on debug. My main issue is 'failed to EGL with glad'. It seems that I can solve this by running in headless mode. However, I am unable to do so for some reason. I tried setting cfg.show_gui = False in datarenderer.py. But I am still running into same problems. Any help would be appreciated!

BTW, really great work!

lukashermann, 2024-01-07:
@qureshinomaan what exactly are you running? The `datarenderer` in `calvin_env` is not being used during training, we just used it to render the dataset once after recording it with teleoperation. During training, it's the rollout callbacks that use the `calvin_env` simulator, so as a quick fix you could disable them during training (they are just used to evaluate the performance during training), however, you would still need to render at one point for the full evaluation after the training is done. (for disabling them, set `~callbacks/rollout` and `~callbacks/rollout_lh` in the command line arguments for the training).
Also, headless rendering is enabled by default for the rollouts / the evaluation. Does your computer have a graphics card? EGL renders on the GPU, so it would fail if you don't have one.

What is the output if you run [this script](https://github.com/mees/calvin_env/blob/main/egl_check/list_egl_options.py) in `calvin_env`

lukashermann, 2024-01-07:
see https://github.com/mees/calvin/issues/59

closed 2024-01-07 by lukashermann

LukeLIN-web, 2025-06-10:
I am using 
```
cmake
wheel
numpy==1.26.4
hydra-core==1.3.2
hydra-colorlog==1.2.0
matplotlib==3.10.1
opencv-python==4.11.0.86
omegaconf==2.3.0
plotly==6.0.0
pyhash
pytorch-lightning==1.8.6
lightning_lite
torch==2.3.1 # torch==1.13.1
torchvision==0.18.1
MulticoreTSNE
gitpython==3.1.44
scipy==1.12.0
sentence-transformers
setuptools==57.5.0
gym==0.26.2
moviepy==1.0.3
tqdm==4.67.1
termcolor==2.5.0
wandb==0.18.7
```