mees/calvin #67: Data storage format

COST-97, 2024-02-01:
Hello:
We want to collect more data based on this dataset.
Can you elaborate on the role of each file in the training and validation files and the tutorial on data collection?
Thank you so much!

lukashermann, 2024-02-08:
Are you planning to record data with teleoperation using a VR headset?

COST-97, 2024-02-17:
Glad to hear from you!
Your further explanation makes the data format clearer.
We are planning to collect data from other robots in simulations in the near future, and we are not considering using VR yet.

COST-97, 2024-02-28:
Hello:
If I'm not mistaken, in the training phase, in all the tasks, the episode length is 64.
But intuitively there should be some differences between the different tasks, right?
How do you deal with this problem?
Thank you!

lukashermann, 2024-02-28:
You probably refer to the language annotations, for which we use our automatic labeling tool with a sequence length of 64. That means we sample random sequences of lengths 64 and check if any task was solved in that interval, in which case that sequence gets a language label. Since most tasks need less than 64 frames to be solved, there are usually some frames at the beginning or at the end of the sequence that are not strictly task-related (for example the locomotion in the direction of the handles or switches). 
During training, we sample sequences of lengths 16 to 32 (always padding to 32). For the vision part of the dataset, we use a sliding window approach over the whole play data (which can be seen as a couple of very long demonstrations). 
For the language part, we still use a sliding window, but restrict it to those 64 frame sequences that were labeled with a language annotation. If you crop 16 frames from a sequence of 64 frames, it is of course possible that the completion of the task is not shown, but this is not a problem.

COST-97, 2024-03-01:
Hello:
The range of the rel_actions is -1~1.
How is normalization done here?
Select the maximum and minimum values in the data for normalization?
Thanks!

lukashermann, 2024-03-01:
For the conversion from absolute metric space to relative normalized actions, the position component (x,y,z) is clipped at the interval [-0.02, 0.02] and the orientation component (euler angles) at [-0.05, 0.05]. After the clipping, we normalize to the range [-1, 1]. This happens in the [calvin_env](https://github.com/mees/calvin_env/blob/main/calvin_env/utils/utils.py#L160) at the time of rendering.

To convert them to back metric space, the position component (x,y,z) is multiplied by 0.02 and the orientation component (euler angles) by 0.05. This happens in calvin_env [here](https://github.com/mees/calvin_env/blob/main/calvin_env/robot/robot.py#L228), when we do a rollout and want to control the robot with relative actions.

lukashermann, 2024-04-24:
I don't know the min and max values for the state observation, you would have to go through the dataset and check.
What we do is normalize the observation to mean 0 and std 1 with [this](https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/utils/compute_proprioception_statistics.py) script. We save the results in the `statistics.yaml` file in the dataset training and validation folder respectively.

 ```
robot_obs:
  - _target_: calvin_agent.utils.transforms.NormalizeVector
    mean: [0.039233, -0.118554,  0.507826,  1.079174, -0.083069,  1.579753,
           0.054622, -0.736859,  1.017769,  1.792879, -2.099604, -0.993738,
           1.790842,  0.586534,  0.095367]
    std:  [0.150769, 0.1104  , 0.06253 , 2.883517, 0.126405, 0.377196,
           0.030152, 0.334392, 0.172714, 0.240513, 0.3842  , 0.198596,
           0.158712, 0.346865, 0.995442]
scene_obs:
  - _target_: calvin_agent.utils.transforms.NormalizeVector
    mean: [0.150934,  0.119917,  0.000239,  0.042049,  0.487755,  0.47448 ,
           0.057482, -0.088074,  0.431237,  0.046034,  0.030599,  0.027333,
           0.062103, -0.092833,  0.430236, -0.054962,  0.019381,  0.096546,
           0.064944, -0.093058,  0.428381,  0.024941,  0.002746, -0.031589]
    std:  [ 0.125757, 0.09654 , 0.002148, 0.041916, 0.49985 , 0.499348,
            0.146225, 0.119266, 0.050408, 1.430807, 0.676023, 2.017468,
            0.142979, 0.113236, 0.049651, 1.545888, 0.3906  , 1.763569,
            0.143077, 0.11546 , 0.050363, 1.514873, 0.431664, 1.860245]
```

If you want to normalize to -1 and 1, just modify the script I linked, but it will take some time to run through the whole dataset.

COST-97, 2024-10-23:
Hello:
I want to collect some data on CALVIN.
The action type should be consistent with ['actions'] in the existing dataset, that is, Absolute cartesian pose.
Specifically, should I record the `action` in `apply_action(self, action)` in _calvin/calvin_env/calvin_env/robot/robot.py_?

```
...
if not len(action) == 3:
action = self.relative_to_absolute(action)
target_ee_pos, target_ee_orn, self.gripper_action = action
# Record action here, right?
```
If not, could you share your code for recording data?
Thank you very much!