mees/calvin #69: TypeError: Error instantiating 'calvin_env.envs.play_table_env.PlayTableSimEnv' : expected str, bytes or os.PathLike object, not NoneType disconnecting id 0 from server

LPY1219, 2024-02-13:
Hi,
    Thanks for your efforts on building such an amazing benchmark! I am trying to use it to validate my RL algorithm, but i met some problems when I tried the RL jupyter you provide in the repository. I can successfully run the script by using google colab, but when i transform it into .py file and install the related packages, it can not work.
I can successfuly load the package by " from calvin_env.envs.play_table_env import PlayTableSimEnv". I do not know why the program will give me the above error when it begins to execute "  env = hydra.utils.instantiate(cfg.env) ". Can you help me ?

lukashermann, 2024-02-13:
Hi, @LPY1219, 
I did what you described and tried to extract the code from the colab to a .py file:

```
import hydra
import numpy as np
import cv2
from hydra import initialize, compose

with initialize(config_path="./calvin_env/conf"):
  cfg = compose(config_name="config_data_collection.yaml", overrides=["cameras=static_and_gripper"])
  cfg.env["use_egl"] = False
  cfg.env["show_gui"] = False
  cfg.env["use_vr"] = False
  cfg.env["use_scene_info"] = True
  print(cfg.env)

env = hydra.utils.instantiate(cfg.env)
observation = env.reset()
#The observation is given as a dictionary with different values
print(observation.keys())
for i in range(5):
  # The action consists in a pose displacement (position and orientation)
  action_displacement = np.random.uniform(low=-1, high=1, size=6)
  # And a binary gripper action, -1 for closing and 1 for oppening
  action_gripper = np.random.choice([-1, 1], size=1)
  action = np.concatenate((action_displacement, action_gripper), axis=-1)
  observation, reward, done, info = env.step(action)
  rgb = env.render(mode="rgb_array")[:,:,::-1]
  cv2.imshow("win", rgb)
  cv2.waitKey(1)
```

It runs without errors on my computer, can you try it yourself?

LPY1219, 2024-02-14:
Thanks for your reply! I have sovled it by adding __init__ file in the "calvin_env" directory. And then the program can successfully find the package.

lukashermann, 2024-02-26:
@LPY1219 Great that you solved your issue! Just out of curiosity, did you follow the installation instructions in our README? `calvin_env` should get installed with pip in the [install.sh](https://github.com/mees/calvin/blob/main/install.sh) script, so you wouldn't neet an `init` file.

initial-h, 2024-08-21:
> @LPY1219 Great that you solved your issue! Just out of curiosity, did you follow the installation instructions in our README? `calvin_env` should get installed with pip in the [install.sh](https://github.com/mees/calvin/blob/main/install.sh) script, so you wouldn't neet an `init` file.

I installed following README, but still need __init__.py to solve the same error.

FutureTwT, 2024-09-05:
Perhaps because the **calvin_env** library is cloned in the current directory, the **import** process looked for the local path instead of the installation path

1425629616, 2024-11-29:
> Hi, @LPY1219, I did what you described and tried to extract the code from the colab to a .py file:
>
> ```
> import hydra
> import numpy as np
> import cv2
> from hydra import initialize, compose
>
> with [...]

I added this __init__.py file in ./calvin_env, but got the following error:
hydra.errors.MissingConfigException: Primary config directory not found.
Check that the config directory '/root/autodl-tmp/calvin/calvin_env/calvin_env/calvin_env/conf' exists and readable

Where can I get the conf? Thanks for your time.

ByteRainx, 2025-04-05:
This problem still exists, can you give me an accurate method to solve it?

2018320637, 2025-04-06:
It seems necessary to modify line 469 in calvin/calvin_env/tacto/tacto/renderer.py: diff = (color.astype(np.float64) - self._background_sim[camera_index]) * 0.5