mees/calvin #71: Simulator

Parvish-Kakarapalli, 2024-04-02:
Kindly, please tell if there is a simulator for CALVIN or the datasets are all we have?

lukashermann, 2024-04-03:
Yes, there is [calvin_env](https://github.com/mees/calvin_env), a simulated environment running on the pybullet physics engine. We use it during the evaluation, have a look at the [code](https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/evaluation/evaluate_policy.py).

Parvish-Kakarapalli, 2024-04-05:
Could you please how to run the simulator independently, and not during the evaluation.

lukashermann, 2024-04-05:
You can run `calvin_env/calvin_env/envs/play_table_env.py` and it will open the pybullet debug viewer where you can see the environment. To interact with the environment, you would have to provide the robot with actions.

Parvish-Kakarapalli, 2024-04-11:
Could you please what does the scene_obs represent in the state space of the environment if it already mentioned elsewhere please redirect me there.

lukashermann, 2024-04-11:
Does [this](https://github.com/mees/calvin/tree/main/dataset#state-observation) answer your question?

Parvish-Kakarapalli, 2024-04-11:
Yeah, thanks a lot.