mees/calvin #79: How to evaluate different Scene

H-ChanY, 2024-05-07:
Hello, 

I want to evalaute my model on your simulator.

But my model doesn't work at Scene D

So I want to try to test at Scene A, B, C

But when I edit the evalute_policy.py that makes impossible scene env. 

For example: if the task is  grap the red block, there is no red block in env image I think it's hide behind of the slider. 

Could you answer that.

Thank you

lukashermann, 2024-05-07:
I think the problem occurs when you convert the initial condition of a test sequence into the concrete object and scene positions in [this](https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/evaluation/evaluate_policy.py#L128) line.
An initial condition is a dictionary that specifies the scene state at the beginning of the 5 step evaluation sequence.
```
{'led': 1,
 'lightbulb': 1,
 'slider': 'left',
 'drawer': 'open',
 'red_block': 'slider_left',
 'blue_block': 'table',
 'pink_block': 'slider_right',
 'grasped': 0}
```
This gets converted to `robot_obs` and `scene_obs` which are then used to reset the environment state.

You can try to tweak the values in [lines 228 to 232](https://github.com/mees/calvin/blob/main/calvin_models/calvin_agent/evaluation/utils.py#L228)  that specify the exact block positions for scene D such that it works for the other scenes.

lukashermann, 2024-05-07:
If you want to quickly test block positions, I think it is easiest if you run 
`python calvin_env/calvin_env/envs/play_table_env.py scene=calvin_scene_C` and you edit the values in the scene config [here](https://github.com/mees/calvin_env/blob/main/conf/scene/calvin_scene_C.yaml#L42). By default, the positions get randomly sampled but you can also put fixed values as in [this conf](https://github.com/mees/calvin_env/blob/main/conf/scene/calvin_scene_D_eval.yaml). Note that despite the name of the latter conf being `calvin_scene_D_eval`, we currently don't use this conf for the evaluation (it's legacy).

You can interact with the pybullet debug viewer and open the slider to see if the block is at the correct position. Probably it's enough if you change the x-dim of the values and leave y and z unchanged.

ecielyang, 2025-05-06:
I want to evaluate model in A, B, C and D separately. Which parameter in which file should I adjust?

I notice that in calvin/dataset/task_D_D/task_D_D/merged_config.yaml, D is set:
```  
objects:
    fixed_objects:
      table:
        file: calvin_table_D/urdf/calvin_table_D.urdf
```
While in calvin/dataset/task_D_D/task_D_D/config.yaml, A is set:
```
scene:
  _target_: vr_env.scene.play_table_scene.PlayTableScene
  _recursive_: false
  name: calvin_scene_A
```