mees/calvin #102: Error in call to target 'calvin_env.envs.play_table_env.PlayTableSimEnv':

wxy301, 2025-03-04:
I have been working with your CALVIN project and encountered an issue where TactileSensor fails to initialize in PlayTableSimEnv. Despite multiple attempts, I have not been able to resolve it, so I would like to seek your guidance.

Issue Description：
When I set show_gui=True to enable visualization, I encounter the following error:

hydra.errors.InstantiationException: 

[image]

InstantiationException("Error in call to target 'calvin_env.camera.tactile_sensor.TactileSensor':\nAssertionError()\nfull_key: cameras.tactile")
I have attached a screenshot of the full error traceback for reference.

What I Have Tried：
Checked config_path to ensure TactileSensor is reading the correct configuration file.
Adjusted OpenGL/EGL settings, trying glfw, glx, and osmesa for rendering, but the issue persists.
Tested tacto.Sensor() separately, and it initializes correctly in an isolated script, but it fails when used in the CALVIN environment.
 
I would like to ask for your guidance on how to properly initialize TactileSensor while enabling GUI rendering. If there are any known dependencies, environment variables, or specific configurations required, I would greatly appreciate your insights.

Thank you for your time and for sharing this excellent research work. I look forward to your response.

ByteRainx, 2025-04-05:
Have you solved this problem? I also encountered this problem.

chenkang455, 2025-11-10:
Try comment it?